matth-x/MicroOcpp #115: message Tx / Rx  methsod ...help me

jpily, 2023-02-10:
Dear Matt-x ..
Hello. I developed a Ev-charger.I am using Arduino IDE (Platform IO). I am using esp32, and I found the ocpp library and tried to test it, so I built the websocket server through node js and react the connection worked normally.
I would like to test-roll the message transmission and reception, but there is no way to send and receive data, so I would appreciate it if you could share the example code. thank you  ^.^ (0_0) help me plz

================================ serial log===========================
[main] Wait for WiFi: .
WiFi connected
IP address:
192.168.0.16
 connected!
[AO] info (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /opstore.cnf
[AO] info (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /txstore.jsn
[AO] info (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /arduino-ocpp.cnf
[AO] info (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /volatile
[AO] info (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:34): Connected to url: ws://127.0.0.1:8000
[AO] warning (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppConnection.cpp:138): Invalid OCPP message! (though JSON has successfully been deserialized)
[AO] warning (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:38): Processing WebSocket input event failed
[AO] warning (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppConnection.cpp:138): Invalid OCPP message! (though JSON has successfully been deserialized)
[AO] warning (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:38): Processing WebSocket input event failed
[AO] warning (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppConnection.cpp:138): Invalid OCPP message! (though JSON has successfully been deserialized)
[AO] warning (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:38): Processing WebSocket input event failed
[AO] warning (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppConnection.cpp:138): Invalid OCPP message! (though JSON has successfully been deserialized)
[AO] warning (.pio/libdeps/esp32-s3-devkitc-1/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:38): Processing WebSocket input event failed

matth-x, 2023-02-10:
Hi @jpily 

If you initialize the WebSocket library like in [this example](https://github.com/matth-x/ArduinoOcpp/blob/master/examples/ESP-TLS/main.cpp), then you have a direct access to the WebSocket object and can use its `sendTXT` function for any data.

To read what's coming from the server, you can set the build flag for printing out all OCPP traffic to the console. Add

```
build_flags =
    -DAO_TRAFFIC_OUT   ; print the OCPP communication to the serial monitor
```

to the platformio.ini

jpily, 2023-02-10:
Thank you for your reply.
It's Friday in Korean time, so I'll test it on Monday.
Have a nice weekend.^^

closed 2023-08-25 by matth-x