matth-x/MicroOcpp #117: I have a question.I'd appreciate it if you could answe

jpily, 2023-02-14:
Dear Math-x,
Hello? Thank you for your answer.

I have a few questions.

Here's the first question. There are four files in the test folder in the library folder, and I wonder what each test is for. Are these files clients? Or is the code running on the websocket server?

I don't know much about the ocpp1.6 protocol.
So I'm studying the information that constitutes json. It's not easy to implement the ocpp1.6 protocol, so I want to implement an electric charger with a great library that you created. So, I would like to test whether this library works well with the ocpp1.6 protocol applied, but the websocket server for Test is not implemented yet, so could you provide a websocket server for you to test the OCPP library? I would like to test whether the library you created works well. I'd appreciate your help.

matth-x, 2023-02-14:
The test-folder contains programs which execute lots of OCPP operations automatically and check if the results are plausible. This allows to see immediately if a new change to the library broke the base behavior, without doing manual testing. Actually GitHub itself executes the test files for each commit, but it runs them on an Ubuntu-VM, not a microcontroller.

To test this library, I highly recommend setting up a SteVe server. It allows you to go through every OCPP operation, see what's going on and make a thorough evaluation of my project. Instead of SteVe you can also connect this library to a WebSocket echo server. But SteVe is the far better option.

jpily, 2023-02-15:
Thank you for your answer. I failed to run because I downloaded Steve server you recommended. Could you give me a little help to run Steve server successfully?

My environment is as follows.
1. java 11 jdk oracle install
2. meven install
3. Download Steve Server Library
4. cmd : mvn packge -> build 
5. run cmd : java -jar target/steve.jar -> fail. 
6. 
-----------------------log------------------------------------------------------------------------------------------------------
[INFO ] 2023-02-15 18:23:25,280 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2023-02-15 18:23:25,453 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2023-02-15 18:23:25,465 de.rwth.idsg.steve.Application - Date/time zone of the application is set to UTC. Current date/time: 2023-02-15T09:23:25.457Z
[INFO ] 2023-02-15 18:23:25,618 org.eclipse.jetty.server.Server - jetty-10.0.13; built: 2022-12-07T20:13:20.134Z; git: 1c2636ea05c0ca8de1ffd6ca7f3a98ac084c766d; jvm 19.0.2+7-44
[INFO ] 2023-02-15 18:23:25,751 org.eclipse.jetty.server.handler.ContextHandler.steve - No Spring WebApplicationInitializer types detected on classpath
[INFO ] 2023-02-15 18:23:25,756 org.eclipse.jetty.server.session.DefaultSessionIdManager - Session workerName=node0
[INFO ] 2023-02-15 18:23:25,764 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring root WebApplicationContext
[INFO ] 2023-02-15 18:23:25,765 org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization started
[INFO ] 2023-02-15 18:23:26,497 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
[INFO ] 2023-02-15 18:23:26,800 com.zaxxer.hikari.pool.HikariPool - HikariPool-1 - Added connection com.mysql.cj.jdbc.ConnectionImpl@6b6def36
[INFO ] 2023-02-15 18:23:26,801 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
[INFO ] 2023-02-15 18:23:28,859 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2023-02-15 18:23:29,212 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP12
[INFO ] 2023-02-15 18:23:29,221 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2012/06/}CentralSystemService from class ocpp.cs._2012._06.CentralSystemService
[INFO ] 2023-02-15 18:23:29,259 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP15
[INFO ] 2023-02-15 18:23:29,260 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2015/10/}CentralSystemService from class ocpp.cs._2015._10.CentralSystemService
[INFO ] 2023-02-15 18:23:29,288 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP16
[INFO ] 2023-02-15 18:23:29,289 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2023-02-15 18:23:29,293 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemService
[INFO ] 2023-02-15 18:23:33,840 org.hibernate.validator.internal.util.Version - HV000001: Hibernate Validator 6.1.7.Final
[INFO ] 2023-02-15 18:23:34,104 org.springframework.security.web.DefaultSecurityFilterChain - Will secure any request with [org.springframework.security.web.session.DisableEncodeUrlFilter@1b52f723, org.springframework.security.web.context.request.async.WebAsyncManagerIntegrationFilter@39384ba6, org.springframework.security.web.context.SecurityContextPersistenceFilter@3b6d94fc, org.springframework.security.web.header.HeaderWriterFilter@4d992634, org.springframework.security.web.csrf.CsrfFilter@62c019bb, org.springframework.security.web.authentication.logout.LogoutFilter@570f96b7, org.springframework.security.web.authentication.UsernamePasswordAuthenticationFilter@2116e9cf, org.springframework.security.web.savedrequest.RequestCacheAwareFilter@665fb1d8, org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter@1272f6d9, org.springframework.security.web.authentication.AnonymousAuthenticationFilter@e7529ac, org.springframework.security.web.session.SessionManagementFilter@3a7e13c0, org.springframework.security.web.access.ExceptionTranslationFilter@2c3762c7, org.springframework.security.web.access.intercept.AuthorizationFilter@7475f071]
[INFO ] 2023-02-15 18:23:34,125 org.springframework.security.web.DefaultSecurityFilterChain - Will secure Ant [pattern='/api/'] with [org.springframework.security.web.session.DisableEncodeUrlFilter@5748638c, org.springframework.security.web.context.request.async.WebAsyncManagerIntegrationFilter@7e864a8c, org.springframework.security.web.context.SecurityContextPersistenceFilter@4bba628, org.springframework.security.web.header.HeaderWriterFilter@59b5251d, org.springframework.security.web.authentication.logout.LogoutFilter@5a98983b, de.rwth.idsg.steve.config.SecurityConfiguration$ApiKeyFilter@5e60e639, org.springframework.security.web.savedrequest.RequestCacheAwareFilter@344db28d, org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestF [... cut ...]
at de.rwth.idsg.steve.Application.main(Application.java:64) ~[steve.jar:?]
[INFO ] 2023-02-15 18:23:34,547 org.eclipse.jetty.server.Server - Stopped Server@107ed6fc{STOPPING}[10.0.13,sto=5000]
[INFO ] 2023-02-15 18:23:34,547 org.eclipse.jetty.server.Server - Shutdown Server@107ed6fc{STOPPING}[10.0.13,sto=5000]
[INFO ] 2023-02-15 18:23:34,555 org.eclipse.jetty.server.handler.ContextHandler - Stopped o.e.j.w.WebAppContext@4152d38d{/steve,jar:file:/E:/JDH/01_Products/12_EV_Charger/OCCP_TEST/OCPP_STeveServer/steve-master/steve-master/target/steve.jar!/webapp,STOPPED}

jpily, 2023-02-15:
We have successfully started the Steve server you recommended. Thank you.
Now, I have a question for the OCPP test, so I ask the following question.

My environment is as follows.

java 11 jdk oracle install
meven install
Download Steve Server Library
cmd : mvn packge -> build
run cmd:java -jar target/steve.js

  In the ocpp library, you need to define the following

#define OCPP_HOST " ??"
#define OCPP_PORT 8000
#define OCPP_URL "" ?
I'm running Steve's server, so I don't know how to write it.
I tried a few tests, but it was disconnected, so I have a question.

Here are the properties of Steve's server:
server.host = 127.0.0.1
http.port = 8000
https.port = 8443

This is the http address when running Steve's server.
1) Login screen address: http://127.0.0.1:8000/steve/manager/signin
2) Home address after login: http://127.0.0.1:8000/steve/manager/home

matth-x, 2023-02-15:
Hi, good to hear that it works!

First, you need to create a charge point in SteVe and give it a name, the so-called chargeBoxId. For example `esp-charger`. Then SteVe should be set up.

On the ESP, you can comment out the following lines:

https://github.com/matth-x/ArduinoOcpp/blob/27ce8b15cd5158e1b8d637b67fe55416093bfe57/examples/ESP/main.cpp#L25-L30

Replace the my.instance.com with your IP address and adapt the port number. The first connection can be a bit tricky, it will probably require a bit of trial and error. Good luck!

closed 2023-08-25 by matth-x