matth-x/MicroOcpp #119: How to send StatusNotification from main.cpp code, I want to make device Available and Unavailable as per the field conditions

piyush9999, 2023-02-18:
I want to send a Statusnotification to make the charging point "Unavailable" whenever there is powerfailure happens

TDVSVI, 2023-02-20:
me too. Please tell me what should I do.

matth-x, 2023-02-20:
Hi, one way is to set the charger Faulted while the failure persists. Both Faulted and Unavailable are "inoperative" states.

```cpp
//in setup()

    addErrorCodeInput([] () {
        if (check_power_failure()) {
            return "UnderVoltage"; //e.g. if power failure is caused by voltage drop
        }
        return (const char*) NULL;
    });
```

The Unavailable state is normally set by the OCPP backend or by specific events which are covered inside the library. If you need to force the charger into Unavailable though, use the following code snippet:

```cpp
//include section of the source file
#include <ArduinoOcpp/Core/OcppEngine.h>
#include <ArduinoOcpp/Core/OcppModel.h>
#include <ArduinoOcpp/Tasks/ChargePointStatus/ChargePointStatusService.h>
```

```cpp
//when the charger should be forced in Unavailable state
    getOcppEngine()->getOcppModel().getConnectorStatus(0)->setAvailabilityVolatile(false);
```

Please reopen if this doesn't solve the problem.

closed 2023-02-20 by matth-x

TDVSVI, 2023-02-20:
Thank you, And can I set status to preparing?

TDVSVI, 2023-02-20:
> Hi, one way is to set the charger Faulted while the failure persists. Both Faulted and Unavailable are "inoperative" states.
> 
> ```c++
> //in setup()
> 
>     addErrorCodeInput([] () {
>         if (check_power_failure()) {
>             return "UnderVoltage"; //e.g. if power failure is caused by voltage drop
>         }
>         return (const char*) NULL;
>     });
> ```
> 
> The Unavailable state is normally set by the OCPP backend or by specific events which are covered inside the library. If you need to force the charger into Unavailable though, use the following code snippet:
> 
> ```c++
> //include section of the source file
> #include <ArduinoOcpp/Core/OcppEngine.h>
> #include <ArduinoOcpp/Core/OcppModel.h>
> #include <ArduinoOcpp/Tasks/ChargePointStatus/ChargePointStatusService.h>
> ```
> 
> ```c++
> //when the charger should be forced in Unavailable state
>     getOcppEngine()->getOcppModel().getConnectorStatus(0)->setAvailabilityVolatile(false);
> ```
> 
> Please reopen if this doesn't solve the problem.

Thank you, And can I set status to preparing?