matth-x/MicroOcpp #121: Controller forgets the OCPP URL after power recycle

mukeshtalks, 2023-02-22:
@matth-x  I am facing an issue with the code attached in this ticket. Controller forgets the OCPP url after power recycle. Can you please help where I am doing wrong? 
[main.txt](https://github.com/matth-x/ArduinoOcpp/files/10803761/main.txt)

mukeshtalks, 2023-02-27:
@matth-x can you please have look at this and help me.

matth-x, 2023-02-27:
Hi @mukeshtalks, sorry for the delayed response. Can you capture the Serial output and post the part where you set the URL?

mukeshtalks, 2023-03-13:
@matth-x please help me i have shared the log in my last post and way i am configuring the webs socket URL

matth-x, 2023-03-14:
The device loses the configuration between the boot sequences. Everything looks okay at the first boot and at the second boot, so the error could originate in between. Can you search for more debug messages (starting with `[AO] ERROR `)? It could, for example complain about flash memory shortages or other file system errors which could damage the URL store.

Generally, the SECC example was just intended for proof-of-concepts. After all, the integration of WiFiManager is not the best and I would advise to write an own maintenance portal for productive use (WiFiManager or another project, at your choice).

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x