matth-x/MicroOcpp #124: lte module

jpily, 2023-03-13:
Dear Math-x 
I have a question about the library.
Can I use the library using a TCP/IP connection through lte module (sim7600 1lc) without using a wifi connection?
There is an infinite reboot

matth-x, 2023-03-14:
Hi @jpily 

Do you mean the route of the bootNotification through the library? It is:

- In ArduinoOcpp.cpp, `bootNotification` calls `ocppEngine->initiateOperation(std::move(bootNotification));`
- OcppEngine forwards it to OcppConnection via `oConn.initiateOcppOperation(std::move(op));`
- OcppConnection queues initated operations and sends them out delayed in another loop call. Eventually it calls `sendReq(ocppSock);` of the operation
- OcppOperation serializes the BootNotification message using `createReq()` and sends it using `ocppSocket.sendTXT(out);`

jpily, 2023-03-15:
You must be busy until late, but thank you for taking the time for your sincere reply. Ask for additional help.
1. I want to create a separate function in main.cpp to create Tx and Rx OCPP-J (json format) messages and test them directly on Steve-server. You need to create a transmit/receive function and finally implement the connection with Steve-server without using wifi using TCP/IP using the sim7600 1lc LTE module.

Sending and receiving via AT command is needed, so I would appreciate any advice to create a function.

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x