matth-x/MicroOcpp #126: Add web socket authentication on OCPP_initialize

pedro-fuoco, 2023-03-15:
Hi! 
My application uses web socket authentication when connecting with the Central System, I imagine this is a common security measure in EV chargers.
I was thinking about making a pull request to add "login" and "password" optional arguments to OCPP_initialize, what do you think? @matth-x

pedro-fuoco, 2023-03-15:
Defaulting to SSL might be worth it as well. We can add the fingerprint as an optional argument and just send an empty string if not used.

matth-x, 2023-03-17:
Hi Pedro,

If I'm honest, I never really liked the initialize function myself. Very happy to accept improvements.

To provide the community with the best code, it should be aligned to the most practical requirements. The WebSocket authentication header is a very common practice in OCPP. Fingerprints are wide spread in IoT controllers, but with OCPP I've only seen TLS certificates so far. So is it an option for you to provide a certificate-based initialize function?

I also like the idea to engage people to move from WS to WSS much quicker. It would be a huge improvement to combine both in the same initialize function. Do you think it's possible enable / disable TLS based on the URL?

closed 2023-08-25 by matth-x