matth-x/MicroOcpp #129: Dear matth-x  Could you tell me how to use the function? I'm not sure about the factor composition

jpily, 2023-03-16:
Dear matth-x
Did you have a good day?
I'm investing a lot in analysis with the great library you made.
1. viod OCPP_initialize()
2. viod OCPP_deinitialize()
3. void OCPP_loop()
4. void bootNotification()
5. void authorize()
6. bool beginTransaction()
7. bool endTransaction()
8. bool isTransactionRunning()
9. bool ocppPermitsCharge()
10. void setConnectorPluggedInput()
11. void setEnergyMeterInput()
12. void setPowerMeterInput()
13. void setSmartChargingOutput()
14. void setEvReadyInput()
15. void setEvseReadyInput()
16. void addErrorCodeInputEV()
17. void addMeterValueInput()
18. void setOnResetNotify()
19. void setOnResetExecute()
20. void setOnUnlockConnectorInOut():
21. void setConnectorLockInOut():
22. void setTxBasedMeterInOut()
23. bool isOperative()
24. int getTransactionId()
25. const char *getTransactionIdTag()
26. bool startTransaction()
27. bool stopTransaction()

I'm not sure how to use these functions in main.cpp in a defined function. Unfortunately, the library is written based on c++, so I don't have any knowledge.
I want to send the meter value to the Steve server and check the meter value on the Steve server.
My prediction is that you can use the addMeterValueInput()

I'm not sure how to send the addMeterValueInput(????) factors.

As defined in ArduinoOcpp.cpp,

void addMeterValueInput(std::function<int32_t ()> valueInput, const char *measurand, const char *unit, const char *location, const char *phase, uint connectorId)

That's how the factor composition works. I would appreciate it if you could tell me how to use this to send it.

While searching, addMeterValueInput([]{return120;}, "Voltage", "V", nullptr, nullptr); I wrote it like this, but nothing was sent to Steve server.

It would be nice to have an example of how to use each function.
I hope you have a happy day today.

matth-x, 2023-03-17:
Hi @jpily 

By default the library only sends Meter Values during a transaction. So the first thing to get started is to start and stop transactions. You can use the function `beginTransaction(idTag)` for that. Also see this: https://github.com/matth-x/ArduinoOcpp/issues/128

Your example code of the Voltage reader seems good. You still need to enable voltage reading on OCPP side which is done by the Configuration keys. You can set them in SteVe. Adding `Voltage` to `MeterValuesSampledData` should do the trick.

jpily, 2023-03-20:
I don't know how to send the meter value.
This is my code. I am asking for help me

I don't know where to set up the Steve server. Looking for data...

#include <Arduino.h>
#include <WiFi.h>
#include "lte.h"
#include "OcppMessage.h"
#define PIN 4

static CR95HF reader(15, 2, -1, -1);
uint32_t preTime = 0;

void setup()
{
   Serial. begin(SERIAL_BAUDRATE);

   // startModemSerial();
   WiFi.begin(STASSID, STAPSK);
   while (WiFi.status() != WL_CONNECTED)
   {
     delay(500);
     Serial.print(".");
   }
   Serial. println(F(" connected!"));
   delay(1000);
   Serial. println("");

   reader.begin();
   Serial. println(reader. readSerial());

   Serial.println("Connecting WebSocket");
   Serial.println("1. OCPP_initialize initialization and connection");
   OCPP_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL);

   Serial.println("2. bootNotification: Sends a signal to the charging station that the charger has finished booting");
   bootNotification("needs charger", "needsrobot office");
}

void loop()
{
     // set pixel to red, delay(1000)
   OCPP_loop();
   authorizeFunction(reader);
   MeterValueInput();
}

================================================== ==
#include "OcppMessage.h"

void authorizeFunction(CR95HF& reader)
{
     if (!getTransactionIdTag()) // Parameter is connecter id (if there are 10 charger cables, id 0 ~ 10 is given)
     {
         Serial.println("Tag Read start test");
         String idTag = reader. getID();
         Serial. println(idTag);
        
         authorize(idTag.c_str(), [idTag](JsonObject response) { // send authentication message
        
             if (response["idTagInfo"]["status"] == "Accepted")
             {

                 Serial.println("1. User authentication complete");
                 bool res = beginTransaction(idTag.c_str());
                 if (res == true)
                 {
                     Serial.println("2.Transaction start and charge progress");
                    
                 }
                 else
                 {
                     Serial.println("Transaction failed.");
                 }
             }
         });
     }
}

void MeterValueInput() {
     if(getTransactionIdTag()) {
         Serial.println("3. Check transaction in use");
     addMeterValueInput([] () {return (25);}, "Current.Import","A");

     }

}

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x