matth-x/MicroOcpp #130: How I can get data information of data transfer from OCPP server?

TDVSVI, 2023-03-17:
Hello everyone, Can you help me about Data Transfer from OCPP server?
I would like to get data and information on main.cpp. I have to use this data to control my EV Charger.

Thank you.

jpily, 2023-03-17:
platformio.ini

build_flags = 
	-D AO_DBG_LEVEL=AO_DL_INFO
	-DAO_TRAFFIC_OUT
	-D ARDUINOJSON_ENABLE_STD_STRING=1

==========================================
Platform because it is a library created based on the ocpp1.6 protocol scenario.Ini.

build_flags =
-D AO_DBG_LEVEL=AO_DL_INFO
-DAO_TRAFFIC_OUT
-D ARDUINOJSON_ENABLE_STD_STRING=1
Additionally, download the firmware again and test it to see the log of the message transmission and reception process.

matth-x, 2023-03-17:
Hi @TDVSVI,

Yes, that's possible. See this https://github.com/matth-x/ArduinoOcpp/issues/93#issuecomment-1286880859 for a guide. 

Please reopen for follow-up questions.

closed 2023-03-17 by matth-x