matth-x/MicroOcpp #132: How to use the TLS example and how to connect to WSS url

piyush9999, 2023-03-27:
Hey @matth-x I have worked on this Libarary till now using normal unsecure connection i.e with port number 8180 and 8080, Just last week i started working on the WSS urls and my device is not getting connected 

I'm getting error as "Certificate verification failed, e.g. CRL, CA or signature check failed'

What changes should i consider while working with Secure websocket urls

pedro-fuoco, 2023-03-28:
#127 will solve your issue, you can look into the commits I've made or just wait for it to be merged into master (this should happen soon)

pedro-fuoco, 2023-03-28:
@matth-x great call about using TLS instead of SSL! I'm happy that the PR will help @piyush9999

piyush9999, 2023-03-28:
Hey @pedro-fuoco yes, not so urgent for me even, I am also trying that. Thank you

matth-x, 2023-08-25:
#127 has been merged and solves this issue.

closed 2023-08-25 by matth-x