matth-x/MicroOcpp #133: Add idTag in StopTransaction.req

dipankarkhartad, 2023-03-28:
Hi @matth-x 

Thank you very much for this library, I am working on this library from past few days and you and team are very support. I have solved many problems with your help.

**I want to add idTag in the StopTransaction.req message, is there any way to add it?**

Kindly refer my StopTransaction.req message.

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Tasks/ChargePointStatus/ConnectorStatus.cpp:338): Session mngt: trigger StopTransaction

[AO] Send: [2,"1000012","StopTransaction",{"meterStop":2310,"timestamp":"2023-03-28T04:47:37.000Z","transactionId":387801,"transactionData":[{"timestamp":"2023-03-28T04:41:35.000Z","sampledValue":[{"value":"1644","context":"Transaction.Begin","measurand":"Energy.Active.Import.Register","unit":"Wh"}]},{"timestamp":"2023-03-28T04:47:38.000Z","sampledValue":[{"value":"2421","context":"Transaction.End","measurand":"Energy.Active.Import.Register","unit":"Wh"}]}]}]

jpily, 2023-03-28:
Would it be appropriate to include the Tag ID when sending in JSON format?

dipankarkhartad, 2023-03-28:
According to OCPP document, 
[image]

The idTag is optional, CP can add it. 

What is your opinion?

jpily, 2023-03-28:
Using TagId may be appropriate for determining a user's charging status and user classification, but if you are considering deploying your solution for service or obtaining OCPP1.6 certification after development, and if you assume that socket communication will be used with an LTE module, it is advisable to minimize unnecessary data transmission

dipankarkhartad, 2023-03-29:
@jpily 

Thank you for suggestion.

closed 2023-03-29 by dipankarkhartad