matth-x/MicroOcpp #136: Not able to get the errorCode and vender error code when I send the "Trigger Massages->->Status Notification" command from CMS

npd04, 2023-04-05:
Hi **@matth-x**,
Everything is working fine in your nice ocpp librery. Thanks for that.
I had a strange problem. If there is an occurrence of any fault like "**OverVoltage**", I arrange the code like this in setup():-

    addErrorCodeInput([] () {
      if (voltage>260){return "OverVoltage";}
      return (const char*) NULL;
    });

I got the errorCode only once when the error arrives-
[AO] Send: [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"OverVoltage","status":"Faulted","timestamp":"2023-04-05T10:19:38.000Z"}]
[AO] Recv: [3,"1000002",{}]

But after that, in the next cycle of StatusNotification or StatusNotification from CMS, I got this only
[AO] Send: [2,"1000003","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Faulted","timestamp":"2023-04-05T10:20:15.000Z"}]
[AO] Recv: [3,"1000003",{}]

I got the status of "Faulted" but errorCode "NoError" instead of OverVoltage.
Kindly help me to get rid of it and please also give some hints about how to add vender error code in StatusNotification.
Your help is highly appreciated.
Thanks.

matth-x, 2023-04-29:
Hi @npd04,

Good point, that's an incorrect behavior of the OCPP lib and I didn't see it when testing. Will resolve this and continue this thread once there are updates.

matth-x, 2023-08-25:
This issue is resolved starting from version 0.3.0

closed 2023-08-25 by matth-x