matth-x/MicroOcpp #137: ESP32 rebooting after conection with server

GabrielAtlasPower, 2023-04-19:
Hi. Thank you Matth for the great library available here.

I'm having an issue with an ESP32 when it connects with the backend server. The debug text below shows a possible problem. Do you know what kind of problem is that and had experienced that before?

_[AO] Send: [2,"1000000","BootNotification",{"chargePointModel":"My Charging Station","chargePointVendor":"My company name"}]
[AO] Recv: [2,"6a6b165d-bb4e-425a-871f-6e524a50f651","GetConfiguration",{}]
[AO] Send: [3,"6a6b165d-bb4e-425a-871f-6e524a50f651",{"configurationKey":[{"key":"ConnectionTimeOut","readonly":false,"value":"30"},{"key":"MinimumStatusDuration","readonly":false,"value":"0"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"true"},{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"false"},{"key":"LocalPreAuthorize","readonly":false,"value":"false"},{"key":"AO_SilentOfflineTransactions","readonly":false,"value":"false"},{"key":"AO_FreeVendActive","readonly":false,"value":"false"},{"key":"AO_FreeVendIdTag","readonly":false,"value":""},{"key":"ResetRetries","readonly":false,"value":"2"},{"key":"HeartbeatInterval","readonly":false,"value":"300"},{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"AO_MeterValueCacheSize","readonly":false,"value":"1"},{"key":"MeterValueSampleInterval","readonly":false,"value":"60"},{"key":"StopTxnSampledData","readonly":false,"value":""},{"key":"MeterValuesAlignedData","readonly":false,"value":"Energy.Active.Import.Register,Power.Active.Import"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"StopTxnAlignedData","readonly":false,"value":""},{"key":"AO_MeterValuesInTxOnly","readonly":false,"value":"true"},{"key":"AO_StopTxnDataCapturePeriodic","readonly":false,"value":"false"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,RemoteTrigger,FirmwareManagement,SmartCharging"},{"key":"AuthorizeRemoteTxRequests","readonly":true,"value":"false"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"30"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"8"},{"key":"StopTxnSampledDataMaxLength","readonly":true,"value":"8"},{"key":"MeterValuesAlignedDataMaxLength","readonly":true,"value":"8"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"8"},{"key":"ChargingScheduleAllowedChargingRateUnit ","readonly":true,"value":"Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"24"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"10"}]}]
[AO] Recv: [3,"1000000",{"status":"Accepted","currentTime":"2023-04-19T01:33:59.876Z","interval":300}]
[AO] info (src/ArduinoOcpp/MessagesV16/BootNotification.cpp:94): Request has been accepted
[AO] debug (src/ArduinoOcpp/Core/FilesystemAdapter.cpp:139): File open successful: /op-122.jsn
[AO] debug (src/ArduinoOcpp/Core/FilesystemUtils.cpp:72): Loaded JSON file: /op-122.jsn
[AO] debug (src/ArduinoOcpp/Core/OcppOperation.cpp:373): restore unique_id_counter with 1000197
[AO] debug (src/ArduinoOcpp/Tasks/Transactions/TransactionStore.cpp:85): 1-61 does not exist
[AO] ERROR (src/ArduinoOcpp/MessagesV16/StartTransaction.cpp:106): referential integrity violation
[AO] ERROR (src/ArduinoOcpp/Core/OcppOperation.cpp:404): restore opNr 122 error
[AO] ERROR (src/ArduinoOcpp/Core/OperationsQueue.cpp:87): could not restore operation
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40196a57  PS      : 0x00060a30  A0      : 0x800df339  A1      : 0x3ffb1780
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffd0500  A5      : 0x3ffc6714  
A6      : 0x3ffb173c  A7      : 0x3ffb173c  A8      : 0x800deecd  A9      : 0x3ffb1760
A10     : 0x3ffd0500  A11     : 0x3ffd0278  A12     : 0x00000001  A13     : 0x3ffb1780  
A14     : 0x3ffb1700  A15     : 0x00000008  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x40196a57:0x3ffb1780 0x400df336:0x3ffb17a0 0x400dfd75:0x3ffb1850 0x400d98f7:0x3ffb1930 0x400daec9:0x3ffb1970 0x400db1d1:0x3ffb1b00 0x400dd7f0:0x3ffb1b20 0x401976ef:0x3ffb1b50 0x400f67c1:0x3ffb1b80 0x400f8045:0x3ffb1ba0 0x40197769:0x3ffb1bd0 0x400f82ee:0x3ffb1bf0 0x400f8389:0x3ffb1c20 0x400f85a1:0x3ffb1c40 0x400f862a:0x3ffb1c80 0x400f82ee:0x3ffb1ca0 0x400f8389:0x3ffb1cd0 0x400f83f1:0x3ffb1cf0 0x400f85e2:0x3ffb1d30 0x400f860f:0x3ffb1d70 0x400f7905:0x3ffb1d90 0x400f7b32:0x3ffb1ec0 0x400dd755:0x3ffb1ee0 0x400db219:0x3ffb1f00 0x400d2894:0x3ffb1f20 0x400f469d:0x3ffb1f40 0x40100c8d:0x3ffb1fb0 0x40089a6a:0x3ffb1fd0

Rebooting...
ets Jul 29 2019 12:21:46_

matth-x, 2023-04-19:
Hi @GabrielAtlasPower, Thanks for reporting this!

The issue here is a filesystem corruption. I can see that there is a pending StartTransaction, but the transaction record is missing. I was able to reproduce this scenario and found the issue. de422dfc3767f30043128e6e8d1b2d1b505a2e4c fixes this and corrupt entries will be deleted.

Is it possible that the flash memory is full?

And btw. can you find out which commit you're using?

closed 2023-04-19 by matth-x

GabrielAtlasPower, 2023-04-20:
Hi @matth-x!! 

The flash memory was 92%. I was using the master commit from 2 weeks ago. 
With you fixes the firmware is running here without problems! Thank you for the fast fixes!!