matth-x/MicroOcpp #138: I AM TRYING TO CONNECT ESP32 TO STEVE WEBSERVER USING OCPP1.6. BUT I AM NOT ABLE TO CONNECT IT TO STEVE NOT EVEN THE HEARTBEAT.I GET THE FOLOWING ERROR

rathourcode, 2023-04-26:
[image]

Pablo2048, 2023-04-26:
Why are you yelling? Did you add charger identity in Steve and did you put correct URL into ArduinoOCPP?

ISCAETY, 2023-04-26:
1. Please confirm your steve server is running.
2. Please check host ,port and url of steve server in your code is correctly .
3. if your steve server is wss not ws ,you could modify your code to support it.

rathourcode, 2023-04-26:
> 1. Please confirm your steve server is running.
> 2. Please check host ,port and url of steve server in your code is correctly .
> 3. if your steve server is wss not ws ,you could modify your code to support it.

MY STEVE SERVER IS RUNNING
THE URL GENERATED BY STEVE IS ONE I USED IN CODE EXAMPLE PROVIDED BY MATHEWS
MY STEVE SERVER ID WS

rathourcode, 2023-04-26:
> Why are you yelling? Did you add charger identity in Steve and did you put correct URL into ArduinoOCPP?

YES I HAVE MADE CHARGER IDENTITY IN STEVE AND I HAVE PUT THE WS URL IN CODE

rathourcode, 2023-04-26:
Please help me i have tried everything

ISCAETY, 2023-04-26:
I think you should check your network if steve server is running on a private network or a public network like a firewall. 
May be you can test connect to steve websocket by postman on your computer.

rathourcode, 2023-04-26:
hai,i tested it using postman ,it was unable to connect to my webserver

when i checked steves log it gives ocpp protocol not specified hence denied connection request
what should i do know

Fareed99, 2023-04-26:
This happened with me and it held me for months tbh.
1. First and foremost, make sure that you have set port forwarding as required and mentioned in this link https://github.com/steve-community/steve/wiki/Prepare-Ubuntu-VM-for-SteVe#optional-ufw-configuration under the "(optional)  UFW Configuration" heading.
2. "Connection reset by peer" appears when the SteVe VM is running but the command to run SteVe on the VM is not executed. Make sure you run SteVe using this command.
```
# java -jar target/steve.jar
```
3. In the code for ESP32, mention the host, port and url like so:
```
#define OCPP_HOST "192.168.18.34"
#define OCPP_PORT 8080
#define OCPP_URL "ws://192.168.18.34:8080/steve/websocket/CentralSystemService/TestCharger"
```
make sure to replace the host, port and charger name as per your settings.

rathourcode, 2023-04-28:
hai i was able to coonect to my steve webserver using postman but still my esp32 is not connecting

matth-x, 2023-04-29:
Hi everyone, thanks for sharing your advice on this!

@rathourcode, sorry for jumping in later. In many cases, the connection problems are caused by firewall settings or by incomplete connection details (see Fareed99's comment for a complete example: https://github.com/matth-x/ArduinoOcpp/issues/138#issuecomment-1523264132). 

If Postman connects while running on the same computer as SteVe, the chances are still high that some firewall settings are still missing.

If Postman connects while running on a different computer, it's a hint that something is not going well with the DNS resolver. In that case I would recommend trying it with the pure [WebSockets library](https://github.com/Links2004/arduinoWebSockets). Bypassing the OCPP lib will simplify things a lot for resolving the connection issue. Basically, all my library does is to forward the credentials to the WS lib in that line:
https://github.com/matth-x/ArduinoOcpp/blob/0da4df1aa9b7fce37b8279d88b28f6361ae362e8/src/ArduinoOcpp.cpp#L65
In that setup, you could try exchanging IP addresses with the hostname and so on. Somehow it will work for sure.

By coincidence I saw that you opened an issue in the SteVe project (https://github.com/steve-community/steve/issues/1132). Is it related? If yes, what were your steps to get that result?

rathourcode, 2023-04-29:
> Hi everyone, thanks for sharing your advice on this!
> 
> @rathourcode, sorry for jumping in later. In many cases, the connection problems are caused by firewall settings or by incomplete connection details (see Fareed99's comment for a complete example: [#138 (comment)](https://github.com/matth-x/ArduinoOcpp/issues/138#issuecomment-1523264132)).
> 
> If Postman connects while running on the same computer as SteVe, the chances are still high that some firewall settings are still missing.
> 
> If Postman connects while running on a different computer, it's a hint that something is not going well with the DNS resolver. In that case I would recommend trying it with the pure [WebSockets library](https://github.com/Links2004/arduinoWebSockets). Bypassing the OCPP lib will simplify things a lot for resolving the connection issue. Basically, all my library does is to forward the credentials to the WS lib in that line:
> 
> https://github.com/matth-x/ArduinoOcpp/blob/0da4df1aa9b7fce37b8279d88b28f6361ae362e8/src/ArduinoOcpp.cpp#L65
> 
> 
> In that setup, you could try exchanging IP addresses with the hostname and so on. Somehow it will work for sure.
> By coincidence I saw that you opened an issue in the SteVe project ([steve-community/steve#1132](https://github.com/steve-community/steve/issues/1132)). Is it related? If yes, what were your steps to get that result?

First of all great thanks for your library , i will try to do  what you have suugested . thank you very much man for responding.

rathourcode, 2023-05-03:
> Hi everyone, thanks for sharing your advice on this!
>
> @rathourcode, sorry for jumping in later. In many cases, the connection problems are caused by firewall settings or by incomplete connection [...]

hai 
i tried  to connect to my  steve server from different computer using postman but failed , i blocked all firewalls even then  
i was unable to connect  can you suggest where problem could be

Fareed99, 2023-05-05:
Are you running your SteVe server in a VM?
If yes, then is the network adapter setting configured to 'Bridged'?
By default this setting is set to NAT, which means the VM will be sharing the IP with the host computer. Changing this setting to Bridged assigns a different IP to the VM.

If you already followed the port forwarding step I mentioned in my previous comment, then changing the network adapter setting in the VM will do it for you, as it did for me.

rathourcode, 2023-05-05:
> Are you running your SteVe server in a VM? If yes, then is the network adapter setting configured to 'Bridged'? By default this setting is set to NAT, which means the VM will be sharing the IP with [...]

i am not  running steve on VM , running on my windows os  
i used follwing instructions for for configuring my firwall
https://learn.microsoft.com/en-us/sql/reporting-services/report-server/configure-a-firewall-for-report-server-access?view=sql-server-ver16

still got nothing please suggest something

matth-x, 2023-08-25:
Closing this due to inactivity

closed 2023-08-25 by matth-x