matth-x/MicroOcpp #141: Using C++03 complier

hk-moon, 2023-05-03:
As you know,  Power exectric & EV side is using TI TMS320F28377D chipset. But this chipset complier is using C++03 version. Do you have any idea/plan of this version(c++03) re-compling ?

matth-x, 2023-05-04:
This library uses the C++14 standard at the moment. It provides some very essential features compared to C++03, most importantly lambda expressions and smart pointers (those hugely reduce the risk of memory leaks). While a C++11 back port should be possible, I don't believe that you could make the code ready for C++03 easily.

I'm sorry that I cannot provide further help on this.

closed 2023-05-04 by matth-x

hk-moon, 2023-05-08:
Dear,Thank you for your kinds.

Closed #141 as completed.