matth-x/MicroOcpp #142: Ciclic reboot ESP32-Wrover-E

maxikys, 2023-05-04:
Hello Matt-x. 
First, I want to thank you for the work you have done.
my platform ESP32 WROVER-E
Description of the error: Before the kernel update, the Platform worked fine, when I updated all the drivers, I got a cyclic reboot of ESP 32: 

 **rst:0x3 (SW_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46**

Thank you!

Pablo2048, 2023-05-04:
No one can help you wit such little infos I'm afraid. Try to full erase the chip after change SDK version and use exception decoder.

maxikys, 2023-05-08:
After changing paritions to default scheme, all problems are lost. 
thanks.