matth-x/MicroOcpp #145: Disconnection after 10 hours or more

GabrielAtlasPower, 2023-05-10:
I'm facing a problem after many hours of AOCPP running. The log shows failed DNS and OCPP is disconnected. If I reboot, the firmware runs again normally. There is some way the firmware can reconnect after many failed tries? 

[image]

maxikys, 2023-05-14:
Hello in example **[ArduinoOcpp](https://github.com/matth-x/ArduinoOcpp/tree/master)/[examples](https://github.com/matth-x/ArduinoOcpp/tree/master/examples)/[SECC](https://github.com/matth-x/ArduinoOcpp/tree/master/examples/SECC)
/main.cpp**

there are this code
```C++
if (scheduleReboot > 0 && millis() - reboot_timestamp >= scheduleReboot) {
        ESP.restart();
    }
```
setup **scheduleReboot** and ESP will be rebooting.

it's not very elegant but it will work.

matth-x, 2023-05-15:
This message means that the DNS resolver can't reach the name server anymore. I can imagine that some network nodes between the ESP and the name server reset the connection and the ESP networking driver does not properly reconnect. This would be a hint of a bug in the Arduino networking libraries. Note that neither the DNS resolver nor the raw WebSocket connection are in the scope of this library.

There's a good chance that you can optimize the robustness by fine-tuning the connection settings in the [WebSocket library](https://github.com/Links2004/arduinoWebSockets). To find the best settings, I would advise to run a program with the WS lib only, monitoring how long the connection remains open and then trying different settings. If you get better results with different settings, please share them here, that could help a lot of users!

In any case, the workaround of @maxikys will recover the ESP from its invalid state.

The alternative [Mongoose WebSocket adapter](https://github.com/matth-x/ArduinoOcppMongoose) is slightly more robust in case exchanging the WS lib is an option for you. Please refer to the OpenEVSE project for a usage example.

GabrielAtlasPower, 2023-05-15:
Thanks, guys. 
@maxikys I was using the code you mentioned, but the problem appear anyway. 
@matth-x  I will investigate deeper the problem more and any results I will post here! For now, I used the code below to make sure the reconnection and perhaps it is working.

`void Wifi_disconnected(WiFiEvent_t event, WiFiEventInfo_t info)

{
  Serial.println("Disconnected from WIFI");
  Serial.print("Connection Lost Reason: ");
  Serial.println(info.disconnected.reason);
  Serial.println("Reconnecting...");
  WiFi.begin(STASSID, STAPSK);
}`

matth-x, 2023-08-25:
Closing this due to inactivity. Feel free to reopen for follow-up questions.

closed 2023-08-25 by matth-x