matth-x/MicroOcpp #146: How to change OcppEvseState to SuspendedEV when I unplug?

TDVSVI, 2023-05-12:
Hello, I would like to use the function of StopTransactionOnEVSideDisconnect which impact to ev charger behavior.

If StopTransactionOnEVSideDisconnect is false when I unplug the ev, it should be turned to Suspendedev status. But now it can't.
For now when I unplug the ev, it sends stoptransaction and the status is finished.

What should I do? Please tell me the solution

maxikys, 2023-05-14:
Hello!
use this callback in setup(){....}:
```C++
setConnectorPluggedInput([]() {
   return true;
},connectorId);
```
Algorithm: if you charger is ready you should return **true** 
                   if unready - **false**

TDVSVI, 2023-05-15:
Thank you very much for your answer.

Can you help me to do this OCPP function? 
When it's charging and unplug it. The OCPPstate must be "SuspendedEV" if I set "StopTransactionOnEVSideDisconnect" is false.

[image]

matth-x, 2023-05-15:
Hi @TDVSVI,

One of the most recent commits in the develop branch introduced the support for StopTransactionOnEVSideDisconnect. Unfortunately, in earlier versions the library accepted that config from the server while it did not implement its functionality. Can you upgrade, rerun that test and share the results here?

TDVSVI, 2023-05-16:
Hi, @matth-x 
Thank you very much for your support. 

I downloaded the most recent library but it can't function. 
There are no responses about "SuspendedEV" when I set false to StopTransactionOnEVSideDisconnect.

Could you tell me about the code of this function? Where is the file in the library that stores this function?

TDVSVI, 2023-05-17:
Thank you very much @matth-x 
I use the latest library and it can work.

closed 2023-05-17 by TDVSVI