matth-x/MicroOcpp #151: How to use "setOnRemoteStartTransactionSendConf" on main.cpp?

yosipat, 2023-05-19:
Hi, I would like to know about setOnRemoteStartTransactionSendConf function.

When I send remotestart from ocpp server, Can I read the payload data before charger send to server on main code?
(such as Accepted or Rejected)

I try this code on void setup.

`setOnRemoteStartTransactionSendConf([](JsonObject payload)
                                      {
                                        if (!strcmp(payload["status"], "Accepted"))
                                        {
                                          Serial.println("Accepted");
                                        }
                                        else
                                        {
                                          Serial.println("Rejected");
                                        } });`

I use it on the old library of [ArduinoOcpp] --> It's work
But the most recent library [ArduinoOcppSimulator] --> not work

What should I do?

Quartzer01, 2023-05-22:
I have the same problem since i updated the library. TT

yosipat, 2023-05-24:
Of course, that is the same issue. I will try it.

closed 2023-05-24 by yosipat