matth-x/MicroOcpp #152: I can't get response from my ev send response to ocpp server

ISCAETY, 2023-05-22:
Hi everyone,

I try to i try to get response from my ev but not work because (OnSendConfListener onSendConf) that not work 

void setOnRemoteStartTransactionSendConf(OnSendConfListener onSendConf) {
    if (!ocppEngine) {
        AO_DBG_ERR("OCPP uninitialized"); //please call OCPP_initialize before
        return;
    }
    ocppEngine->getOperationDeserializer().setOnResponse("RemoteStartTransaction", onSendConf);
}
Please recommend me how to use them on main.cpp for get all response from them

matth-x, 2023-05-23:
Hi @ISCAETY 

Thanks for the report, I found a bug related to this. Can you try the newest version and post if it works as expected?

ISCAETY, 2023-05-24:
Thank You!! It's work now :)
And I found another issue when we reserved charger and plug after that  .  charger didn't change status from "reserved" to "preparing".

commit da67488e (2023-05-23): fix #152
closed 2023-07-02 by matth-x