matth-x/MicroOcpp #161: ESP-IDF 5.0.2 compilation bin size is 1.1 Mb

akashdk191, 2023-07-09:
[image]

I just compiled the ArduinoOcpp stack with all dependencies in the ESP-IDF 5.0.2 , the bin size generated was beyond the partition limits. 
Is there a way to optimize the bin output.
The OTA feature will not work with the size of current bin.

matth-x, 2023-07-28:
Hi @akashdk191, my latest size check with -Os resulted in ~160kB for the ArduinoOcpp component and ~4kB for ArduinoOcppMongoose. Please refer to the Espressif guide for binary size optimization: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html#reducing-overall-size

In the partitions.csv that comes with the esp-idf example, the OTA partitions are 1.5MB large which should suit most applications and the 4MB version of the ESP32. Maybe increasing the partition size will also help in your case.

In future, I will put much more focus on keeping / optimizing the binary size while new features come in. I'm aware that the flash size is a bottleneck for many projects.

closed 2023-07-28 by matth-x

akashdk191, 2023-07-31:
@matth-x Thanks for the response, I will check the optimisation techniques that you have mentioned and recompile. Flash bin size is the real challenge if the library needs to be migrated to STM32 controller .
I was remapping the Application layers to STM32, then the lib size of ArduinoOcpp exceeds the flash size on H7 series of microcontrollers too. I will verify too, thanks again