matth-x/MicroOcpp #164: Issue regarding Clear Charging Profile

RamanpreetKaur1998, 2023-07-22:
Hi @matth-x, 

We've been working on Smart Charging model. The setChargingProfile seems to work perfectly fine. But we are facing issues with clearChargingProfile. Whenever clearChargingProfile.req is received, output is random. Sometimes websocket connection breaks, respnse with "unknown" status is transmitted to the server, response with random character is transmitted to server and most of the times the ESP32 reset is executed. 
Not able to understand this ambiguous behaviour. For your reference I've mentioned details below:
IDE : Arduino IDE
Board : ESP32 Dev Module
Partition Scheme : Huge APP (3MB No OTA/1MB SPIFFS)

Let me know if any other information is required at my end. 
Hope to hear from you soon!

matth-x, 2023-07-27:
Hi @RamanpreetKaur1998, to track this down, which version of ArduinoOcpp do you use? The package on the PIO registry lags a little bit behind and doesn't contain the newest updates of the SmartCharging module yet.

And can you capture the Serial debug output? In case you are worried about publishing the logs to the public, you can also send it to me privately via email: matthias A⊤ arduino-ocpp DО⊤ com

RamanpreetKaur1998, 2023-07-28:
Hi @matth-x,
Below are the Serial Debug Output Logs, after receiving clearChargingProfile, ESP32 reset is executed.
Arduino OCPP latest version is being used.
Packages listed are as follows :
Package         Current    Wanted    Latest    Type      Environments
--------------  ---------  --------  --------  --------  -----------------------
espressif32     3.5.0      3.5.0     6.3.2     Platform  esp32-development-board
ArduinoJson     6.19.1     6.19.1    6.21.3    Library   esp32-development-board
LittleFS_esp32  1.0.5      1.0.5     1.0.6     Library   esp32-development-board
WebSockets      2.3.6      2.3.6     2.4.1     Library   esp32-development-board
____________________________________________________________________________________
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[main] Wait for WiFi: ..... connected!
[AO] info (ArduinoOcpp.cpp:376): Added ConnectorPluggedInput. Transaction-management is in auto mode now
[AO] info (Connection.cpp:56): Connected to url: /
[AO] Recv: WS pong
[AO] Send: [2,"1000000","BootNotification",{"chargePointModel":"My Charging Station","chargePointVendor":"My company name"}]
[AO] Recv: [3,"1000000",{"currentTime":"2023-07-28T12:08:16.012387","interval":5,"status":"Accepted"}]
[AO] info (BootNotification.cpp:83): request has been Accepted
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 0)
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 1)
[AO] info (SmartChargingService.cpp:520): New limit for connector 1, scheduled at = 2023-07-28T12:08:16.028Z, nextChange = 2037-01-01T00:00:00.000Z, limit = {340282346638528859811704183484516925440.000000,340282346638528859811704183484516925440.000000,2147483647}
[AO] Send: [2,"1000001","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2023-07-28T12:08:16.015Z"}]
[AO] Recv: [2,"6eebf517-5ab2-4880-aecc-13bb6b91d679","ChangeConfiguration",{"key":"LocalPreAuthorize","value":"true"}]
[AO] Send: [3,"6eebf517-5ab2-4880-aecc-13bb6b91d679",{"status":"Accepted"}]
[AO] Recv: [3,"1000001",{}]
[AO] Send: [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2023-07-28T12:08:16.022Z"}]
[AO] Recv: [3,"1000002",{}]
[AO] Send: [2,"1000003","Heartbeat",{}]
[AO] Recv: WS pong
[AO] Send: [2,"1000003","Heartbeat",{}]
[AO] Recv: [3,"1000003",{"currentTime":"2023-07-28T12:08:31.035754"}]
[AO] Send: [2,"1000004","Heartbeat",{}]
**[AO] Recv: [2,"b76d2519-58c2-4f11-9aa7-7bf30ffeb045","ClearChargingProfile",{}]**
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060230  A0      : 0x800dc54f  A1      : 0x3ffb1e70  
A2      : 0xad19676c  A3      : 0xad196768  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffcd24c  A12     : 0x00000020  A13     : 0x3ffd1e0c
A14     : 0x3ffccb00  A15     : 0x3ffcd310  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xad19676c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400014fd:0x3ffb1e70 0x400dc54c:0x3ffb1e80 0x400df02b:0x3ffb1ee0 0x400d999b:0x3ffb1f50 0x400d30b0:0x3ffb1f70 0x400d1fff:0x3ffb1f90 0x400fe201:0x3ffb1fb0 0x40089a6a:0x3ffb1fd0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
______________________________________________________________________________________

Let me know if any other information is required at my end.

matth-x, 2023-08-03:
@RamanpreetKaur1998 Thanks again for reporting this!

commit f55a0a21 (2023-08-03): remove uninitialized error code (fixes #164)
closed 2023-08-03 by matth-x