matth-x/MicroOcpp #165: Ethernet-OCPP

pamukcu09, 2023-07-24:
Is it possible to activate the library with an ethernet connection and not Wifi?

Fareed99, 2023-07-25:
There was a repository named arduinowebsocketspolymorphic. When used instead of the normal arduinowebsockets library, it provided the option of selecting WiFi, mobile data, or ethernet as connection source.
But I cannot find it on Github now.

pamukcu09, 2023-07-26:
I start ESP32 with ethernet but if WiFi.begin is not called I get an error because tcp/ip stack cannot be started

pamukcu09, 2023-07-26:
My aim is to use this library in a different microcontroller, so I want to run this library with ethernet connection.

Pablo2048, 2023-07-26:
So You don't have to bother with ESP32 then... Just write your own websocket client library for your architecture and make it API compatible with links2004 arduinowebsockets.

pamukcu09, 2023-07-26:
Thank you for the replies.

closed 2023-07-26 by pamukcu09

matth-x, 2023-07-27:
Hi @pamukcu09, the [OpenEVSE](https://github.com/OpenEVSE/ESP32_WiFi_V4.x) project supports the ESP32 ethernet stack for example. The networking hardware is actually a little bit outside of the scope of ArduinoOcpp though.

A further alternative to consider is to start by running the [Mongoose networking library](https://github.com/cesanta/mongoose) on your microcontroller (most likely supported for your platform) and if everything works fine, to use connect it with the OCPP library using the [Mongoose WS adapter](https://github.com/matth-x/ArduinoOcppMongoose).

pamukcu09, 2023-07-31:
Thank you @matth-x . I will try to implement your suggestions.