matth-x/MicroOcpp #166: Find a memory check error

zhushengsheng001, 2023-07-25:
Hi matth-x,
    There is a memory check report by "valgrind --tool=memcheck --leak-check=full --show-leak-kinds=all ..." 
    It says:
`        ==8581== Conditional jump or move depends on uninitialised value(s)
        ==8581==    at 0x1589EE: ArduinoOcpp::Configuration<int>::operator=(int const&) (ConfigurationKeyValue.cpp:137)
        ==8581==    by 0x143DB4: std::shared_ptr<ArduinoOcpp::Configuration<int> > ArduinoOcpp::createConfiguration<int>(char const*, int) (Configuration.cpp:27)   
        ==8581==    by 0x14460E: std::shared_ptr<ArduinoOcpp::Configuration<int> > ArduinoOcpp::declareConfiguration<int>(char const*, int, char const*, bool, bool, bool, bool) (Configuration.cpp:117)
        ==8581==    by 0x1A77CE: ArduinoOcpp::SmartChargingService::SmartChargingService(ArduinoOcpp::Context&, std::shared_ptr<ArduinoOcpp::FilesystemAdapter>, unsigned int) (SmartChargingService.cpp:308)
        ==8581==    by 0x12F0D7: setSmartChargingOutput(std::function<void (float, float, int)>, unsigned int) (ArduinoOcpp.cpp:492)
        ==8581==    by 0x12E9D5: setSmartChargingPowerOutput(std::function<void (float)>, unsigned int) (ArduinoOcpp.cpp:436)
`
   I did some modifications in file Core/ConfigurationKeyValue.cpp, line 136:
   changed
`        initializedValue = true;
        if (value != newVal) {
            value_revision++;
        }
        value = newVal;`
   to 
`        if (initializedValue == true && value != newVal) {
            value_revision++;
        }
        value = newVal;
        initializedValue = true;`

the valgrind output disappeared.

matth-x, 2023-07-27:
Hi @zhushengsheng001, thanks for reporting this and yes, this makes absolute sense.

Do you have further findings? And can you imagine to make a pull request?

matth-x, 2023-08-09:
Fixed in #168.

closed 2023-08-09 by matth-x