matth-x/MicroOcpp #172: Guidance needed on Porting ArduinoOCPP library on STM32 controller

mukeshtalks, 2023-08-08:
Hello All and @matth-x ,

I want to port this library on STM32 Controller. I want to use STM32 CUBE IDE to compile the code need suggestion and guidance about how this can be ported, What are the dependencies of Arduino I need to resolve? If anyone already experienced this or tried please help. Is there any need to RTOS or bare matel can be ported?

mukeshtalks, 2023-08-08:
@matth-x  thank you for your prompt reply. Sure I will try and keep posting in case any issue I face. If I f I understood correctly the same library you have used as a backend in Arduino OCPP Simulator and integrated with the web interface on the Linux environment. And you have used the Arduino Libraries. If I take the same Arduino Simulator source code without a web interface and make the changes as you suggested and compile it on STM32 CUBE IDE  will this help me to start with? If yes then it will be starting point for me.

mukeshtalks, 2023-08-09:
@matth-x thank you for your reply. I will try and keep posting.

closed 2023-08-09 by mukeshtalks

pamukcu09, 2023-08-23:
> @matth-xCevabın için teşekkürler. Paylaşım yapmaya devam edeceğim.

Hi @mukeshtalks, Have you made any progress in integrating the library into STM32?

What should we define MOCPP_PLATFORM as?

GanesanGuru, 2023-11-20:
> * little

Hi @matth-x ,
Regarding the FileSystem Adapter for iMX RT1064,
Can we skip or deactivate the FS completely for initial integration. If so how to do so - as MOcppMongooseClient [constructor](https://github.com/matth-x/MicroOcppMongoose/blob/main/src/MicroOcppMongooseClient.cpp#L26C71-L26C71) requires FilesystemAdapter as parameter..