matth-x/MicroOcpp #174: How to restore previous charging session when power cut off during charging

demirschmid61, 2023-08-14:
Hi @matth-x,

Thank you very much for this beautiful OCPP library, congratulations to you and your team.

I am able to complete one charging cycle remotely as well as locally. The meter values are also sending with the 15 sec sampling time. 

I am facing one problem when the charging is in progress and sudden power fails, then how do I regain the previous charging session? 

The following steps I did for the scenario:

Step 1: Connect the GUN (Preparing)
Step 2: Start Charging (Charging)
Step 3: Sending MeterValues
Step 4: OFF the ESP32 board..

When I turned ON the board the board is not able to communicate with the OCPP server, the error logs is mentioned below.

*****************************************LOG Started**************************************************
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[main] Wait for WiFi: ... connected!
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /arduino-ocpp.cnf

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /opstore.cnf

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /txstore.jsn

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: /volatile

[E][WiFiClient.cpp:485] flush(): fail on fd 54, errno: 11, "No more processes"
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:31): Disconnected

[E][WiFiClient.cpp:485] flush(): fail on fd 55, errno: 11, "No more processes"
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:31): Disconnected

[E][WiFiClient.cpp:485] flush(): fail on fd 56, errno: 11, "No more processes"
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:31): Disconnected

[E][WiFiClient.cpp:485] flush(): fail on fd 57, errno: 11, "No more processes"
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:31): Disconnected

*****************************************LOG END*************************************************

Can you please go through my issue and suggest any solution for the issue.

matth-x, 2023-08-15:
Hi @demirschmid61,

Thanks for the detailed report! Did you use the master or the develop branch and can you find out the download date or the commit hash?

matth-x, 2023-08-16:
This looks like an issue with the WebSocket library integration. There has been quite a lot of work to optimize it. Can you upgrade to the latest version? (i.e. the current code of this GitHub repo)

To answer your questions: it really depends on the target system if a reboot of the ESP should trigger a transaction stop. In commercial systems where the user pays for the electricity this is typically required. When OCPP is just used for supervision, it is not necessary.

In doubt, you can place `endTransaction(nullptr, "PowerLoss");` at the beginning of the program. It will stop a potentially interrupted transaction or do nothing in case no transaction is still active.

demirschmid61, 2023-08-16:
Thank you very much math for your suggestion. 

I have tried with the endTransaction(nullptr, "PowerLoss"); at the beginning of the program. The session is stopped which was stuck due to power loss.

But from a commercial point of view, meterStart value in startTransaction request is available with the OCPP server and it needs the meterStop value from the stopTransaction to calculate total energy consumed.

In my case, the meterStop value is not fetch from the meterValues (before powerloss)     

##########Log Started##########
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:34): Connected to url: /ACTestCharger
[AO] Send: [2,"1000000","BootNotification",{"chargePointModel":"ABCD","chargePointSerialNumber":"01901","chargePointVendor":"Demirschmid","firmwareVersion":"0.0.1","meterSerialNumber":"0.0.1"}]
[AO] Recv: WS pong
[AO] Recv: [3,"1000000",{"currentTime":"2023-08-16T17:55:11.120260300Z","interval":900,"status":"Accepted"}]
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/BootNotification.cpp:94): Request has been accepted
he boot notification accepted flag
[AO] Send: [2,"1000001","BootNotification",{"chargePointModel":"ABCD","chargePointSerialNumber":"01901","chargePointVendor":"Demirschmid","firmwareVersion":"0.0.1","meterSerialNumber":"0.0.1"}]
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Available (connectorId 0)
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Available (connectorId 1)
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Available (connectorId 2)
[AO] Recv: [3,"1000001",{"currentTime":"2023-08-16T17:55:12.135003700Z","interval":900,"status":"Accepted"}]
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/BootNotification.cpp:94): Request has been accepted
The boot notification accepted flag 1
[AO] Send: [2,"1000002","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2023-08-16T17:55:11.000Z"}]
[AO] Recv: [3,"1000002",{}]
[AO] Send: [2,"1000003","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2023-08-16T17:55:11.000Z"}]
[AO] Recv: [3,"1000003",{}]
[AO] Send: [2,"1000004","StatusNotification",{"connectorId":2,"errorCode":"NoError","status":"Available","timestamp":"2023-08-16T17:55:11.000Z"}]
[AO] Recv: [3,"1000004",{}]
[AO] Recv: WS pong
**Your Charger Status:ev_connect**
########EV Connected######
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp.cpp:308): Added ConnectorPluggedSampler. Transaction-management is in auto mode now
0
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Preparing (connectorId 2)
[AO] Send: [2,"1000005","StatusNotification",{"connectorId":2,"errorCode":"NoError","status":"Preparing","timestamp":"2023-08-16T17:55:34.000Z"}]
[AO] Recv: [3,"1000005",{}]
[AO] Recv: WS pong
**Your Charger Status:start_charging**
########Charging Started######
[main] Transaction initiated. StartTransaction will be sent when ConnectorPlugged Input becomes true
[main] Authorizing user with idTag ABCD123
0
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/Tasks/ChargePointStatus/ConnectorStatus.cpp:237): Session mngt: trigger StartTransaction
[AO] Send: [2,"1000006","StartTransaction",{"connectorId":2,"idTag":"ABCD123",**"meterStart":0**,"timestamp":"2023-08-16T17:55:45.000Z"}]
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Charging (connectorId 2)
[AO] Recv: [3,"1000006",{"idTagInfo":{"status":"Accepted"},**"transactionId":939800**}]
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StartTransaction.cpp:145): Request has been accepted
[AO] Send: [2,"1000007","StatusNotification",{"connectorId":2,"errorCode":"NoError","status":"Charging","timestamp":"2023-08-16T17:55:49.000Z"}]
[AO] Recv: [3,"1000007",{}]
[AO] Recv: WS pong
[AO] Recv: WS ping
[AO] Send: [2,"1000008","MeterValues",{"connectorId":2,"transactionId":939800,"meterValue":[{"timestamp":"2023-08-16T17:56:01.000Z","sampledValue":[{"value":"230","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"124","context":"Sample.Periodic","measurand":"Current.import","phase":"L1","unit":"A"},**{"value":"242","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"**},{"value":"50","context":"Sample.Periodic","measurand":"Frequency","unit":"Hz"},{"value":"37","context":"Sample.Periodic","measurand":"Temperature","unit":"Celsius"}]}]}]
[AO] Recv: [3,"1000008",{}]
[AO] Recv: WS pong
[AO] Send: [2,"1000009","MeterValues",{"connectorId":2,"transactionId":939800,"meterValue":[{"timestamp":"2023-08-16T17:56:16.000Z","sampledValue":[{"value":"230","context":"Sample.Periodic","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"124","context":"Sample.Periodic","measurand":"Current.import","phase":"L1","unit":"A"},**{"value":"363","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"}**,{"value":"50","context":"Sample.Periodic","measurand":"Frequency","unit":"Hz"},{"value":"37","context":"Sample.Periodic","measurand":"Temperature","unit":"Celsius"}]}]}]
[AO] Recv: [3,"1000009",{}]
[AO] Recv: WS pong
[... cut ...]
[AO] info (.pio/libdeps/esp32dev/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StopTransaction.cpp:187): Request has been accepted!
[AO] Send: [2,"1000005","StatusNotification",{"connectorId":2,"errorCode":"NoError","status":"Available","timestamp":"2023-08-16T17:58:11.000Z"}]
[AO] Recv: [3,"1000005",{}]
[AO] Recv: WS pong
[AO] Recv: WS pong

##########Log End##########

As per the logs, the user has consumed 726 Wh but, the user will not have to pay any charges. (meterStop(0) - meterStart (0) = 0 kWh). 

**Is there any way to tackle this kind of situation?**

matth-x, 2023-08-18:
I tried to reproduce the first issue (WebSocket connection issue) on the master branch but it works in my case. Probably it has been fixed already and I won't investigate it any further, anyway thanks for the detailed and well commented report.

The second issue (meterStart and meterStop being 0) is probably due to the order of the ArduinoOcpp setup functions. In general, the order is

```cpp
//create OCPP client instance
ocpp_initialize(...);

//set read access to device sensors
setEnergyMeterInput(...); //meterStart and meterStop will be initialized using this Input function

//(...) after setting up available sensors

//stop transaction from previous run (optional)
endTransaction(nullptr, "PowerLoss"); //this function will read the EnergyMeter Input

//(...) in loop():
ocpp_loop();
```

The important thing here is that ArduinoOcpp can read the device sensors only after setting up the corresponding Input functions. And all setup should be done before the program enters the loop stage.

In your logs I can see that the EnergyMeter Input is used correctly, but it's just set a little too late.

demirschmid61, 2023-08-19:
Thank you @matth-x  for your kind reply, 

The websocket is working in my case as well after updating the library. 

For energy meter input, I'll do all setup function in proper way and check.

demirschmid61, 2023-08-19:
Hi @matth-x ,

I followed the setup function as mentioned by you. Used readEnergy function to read dummy whImport values and created dummy states using serial input. 

Is there anything I am missing? Do I need to store meterStop value before power failure or OCPP library is managing it by storing it in the file system?

/******************Order of the Setup Function ****************
OCPP_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL); // OCPP Instance

  setEnergyMeterInput([]()
                      { return readEnergy(); }); // meterStart and meterStop will be initialized using this Input function

  setConnectorPluggedInput([]()
                           { return false; });

  if (isTransactionRunning())
  {
    endTransaction("PowerLoss"); // End transaction if any
    Serial.println(F(" Previous transaction was running now it is closed"));
  }
  else
  {
    Serial.println(F("No previous transaction running.."));
  }
  auto evseDetailsDoc = std::unique_ptr<DynamicJsonDocument>(new DynamicJsonDocument(
      JSON_OBJECT_SIZE(5) + 5 + 5 + 5));
  JsonObject evseDetails = evseDetailsDoc->to<JsonObject>();
  evseDetails["chargePointModel"] = "ABCD";
  evseDetails["chargePointSerialNumber"] = "01901";
  evseDetails["chargePointVendor"] = "Demirschmid";
  evseDetails["firmwareVersion"] = "0.0.1";
  evseDetails["meterSerialNumber"] = "0.0.1";
  bootNotification(std::move(evseDetailsDoc), [](JsonObject response) {}); // bootNotification
/*****************************************/

[ino.txt](https://github.com/matth-x/ArduinoOcpp/files/12386566/ino.txt)
[main.txt](https://github.com/matth-x/ArduinoOcpp/files/12386567/main.txt)

matth-x, 2023-08-19:
This looks correct. However, note that in your example, the reboot also resets the mocked energy meter. When you call `endTransaction(...)` the library reads the EnergyMeterInput at that moment for the `meterStop` value.

demirschmid61, 2023-08-25:
@matth-x,

Is there any way to use the previous(which was stored in flash) meterStop value for endTransaction().

matth-x, 2023-08-25:
This library doesn't save the energy meter reading while the transaction is running. It only stores the values sent with StartTransaction and StopTransaction. The non-volatile energy register needs to be implemented in the firmware code. A possible solution is to store the energy reading every minute in a file and after a blackout, to restore the energy reading before initializing the OCPP lib.

Feel free to reopen if you have follow-up questions.

closed 2023-08-25 by matth-x

demirschmid61, 2023-08-25:
@matth-x 

Thank you very much for your suggestions. It's clear now.