matth-x/MicroOcpp #175: In the latest ArduinoOcpp lib version=0.3.0, how to know if the bootNotification is sent successfully and cp is accepted by backend?

anupamsaikia, 2023-08-20:


matth-x, 2023-08-23:
@anupamsaikia

The BootNotification is sent as soon as the WebSocket connection is established. This can be checked via the Connection object. To get a reference to the Connection, you need to instantiate it explicitly like in the ESP-TLS example:

https://github.com/matth-x/MicroOcpp/blob/894a2071bd0eec96c26647e6927ccc851b4c8052/examples/ESP-TLS/main.cpp#L68-L69

https://github.com/matth-x/MicroOcpp/blob/894a2071bd0eec96c26647e6927ccc851b4c8052/examples/ESP-TLS/main.cpp#L113-L121

Then `osockSecure.getLastConnected()` will return a value > 0 once the connection has been established.

When the BootNotification is accepted, then the charger becomes operative. You can check that with this function:

https://github.com/matth-x/MicroOcpp/blob/894a2071bd0eec96c26647e6927ccc851b4c8052/src/ArduinoOcpp.h#L276

Does this method provide enough information in your scenario?

anupamsaikia, 2023-08-23:
Thank you for your reply @matth-x 
This helps.

closed 2023-08-23 by anupamsaikia