matth-x/MicroOcpp #178: Not able to resend status notifications after disconnected

demirschmid61, 2023-08-24:
Hi @matth-x, 

I came across one observation, please have a look at the scenario. 

1. ESP started
2. Connection established to server via identifier
3. BootNotification sent
4. Status notification sent
5. After some time client disconnected from server
6. After connecting disconnecting multiple times, client get connected to server. 
7. Status notifications for all the connector are not sent

The websocket connection is established properly but status notifications are not sent. 

All the connectors are offline even websocket connection is established properly. 

Can you tell me what could be the problem?

demirschmid61, 2023-08-25:
Hello @matth-x 
Please have a look at the logs.

/***********Log Started***********/
[main] Wait for WiFi: .. connected!  **WiFi Connected**
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: 
/arduino-ocpp.cnf

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: 
/opstore.cnf

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: 
/txstore.jsn

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/Configuration.cpp:96): init new configurations container: 
/volatile

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp.cpp:308): Added ConnectorPluggedSampler. Transaction-management is in auto mode now

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp.cpp:308): Added ConnectorPluggedSampler. Transaction-management is in auto mode now

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:34): Connected to url: /ACTestCharger
**BootNotification Sent**
[AO] Send: [2,"1000000","BootNotification",{"chargePointModel":"S1-01","chargePointSerialNumber":"DK-01901","chargePointVendor":"Demirschmid","firmwareVersion":"0.0.1","meterSerialNumber":"0.0.1"}]
[AO] Recv: [3,"1000000",{"currentTime":"2023-08-25T04:30:50.596882400Z","interval":900,"status":"Accepted"}]
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/BootNotification.cpp:94): Request has been accepted
he boot notification accepted flag
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Available (connectorId 0)

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Preparing (connectorId 1)

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/MessagesV16/StatusNotification.cpp:50): New status: Preparing (connectorId 2)

**StatusNotification Sent**
[AO] Send: [2,"1000001","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2023-08-25T04:30:50.000Z"}]
[AO] Recv: WS pong
[AO] Recv: [3,"1000001",{}]
[AO] Send: [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2023-08-25T04:30:50.000Z"}] 
[AO] Recv: [3,"1000002",{}]
[AO] Send: [2,"1000003","StatusNotification",{"connectorId":2,"errorCode":"NoError","status":"Preparing","timestamp":"2023-08-25T04:30:50.000Z"}]
[AO] Recv: [3,"1000003",{}]

[AO] Recv: WS ping
[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS ping
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:31): Disconnected

// **Here the client is disconnected** 
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:34): Connected to url: /ACTestCharger

**Here the client reconnected  but the status notifications is not sent by the client**  

[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS pong
[AO] Recv: WS ping
[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:31): Disconnected

[AO] info (.pio/libdeps/esp32-development-board/ArduinoOcpp/src/ArduinoOcpp/Core/OcppSocket.cpp:34): Connected to url: /ACTestCharger

[AO] Recv: WS pong
[AO] Recv: WS pong

@matth-x  can you please comment on ..
1. Do I need to send status notification by any function?
2. Is there any function by using which i can send connector status directly?

matth-x, 2023-08-25:
Hi @demirschmid61, no StatusNotification is required after the WebSocket reconnects. Only if the charger status changed during that period.

Does your OCPP backend require this? In case I get more feedback like this I will consider resending a StatusNotification in that scenario.

If you want to manually send a new StatusNotification, you can take the authorize() function of ArduinoOcpp.cpp as an example. I'm not sure which version of this library you are using, so I cannot give an exact code snippet.

https://github.com/matth-x/MicroOcpp/blob/894a2071bd0eec96c26647e6927ccc851b4c8052/src/ArduinoOcpp.cpp#L804-L818

Copy these lines of your ArduinoOcpp version, replace the Authorize class with StatusNotification and change its parameters correspondingly.

demirschmid61, 2023-08-25:
Hi @matth-x ,

My OCPP backend requires the status notification if the WebSocket reconnects, I am using version "0.3.0". Can you share that code snippet of the same?

matth-x, 2023-08-25:
This is an example for version 0.3.0:

```cpp
//necessary includes
#include <ArduinoOcpp/Core/Context.h>
#include <ArduinoOcpp/Model/Model.h>
#include <ArduinoOcpp/Model/ConnectorBase/Connector.h>
#include <ArduinoOcpp/Core/SimpleRequestFactory.h>
#include <ArduinoOcpp/Operations/StatusNotification.h>
```

```cpp
//in your reconnect handler

auto context = getOcppContext();

unsigned int connectorId = 0; //iterate this variable for all connectors
auto connector = context->getModel().getConnector(connectorId);

//create a StatusNotification message containing the connector status and current time
auto statusNotification = ArduinoOcpp::makeRequest(new ArduinoOcpp::Ocpp16::StatusNotification(
       connectorId,
       connector->getStatus(),
       context->getModel().getClock().now());

statusNotification->setTimeout(60000);

context->initiateRequest(std::move(statusNotification));
```

demirschmid61, 2023-08-25:
Thank you @matth-x ,

Is there any ConnectorBase directory?

I could not find it in the library.

demirschmid61, 2023-08-25:
Hi @matth-x ,

Thank you very much for your quick acknowledgment. 

I am approaching towards my objective of creating an OCPP client for two connectors. 

Can you provide any references so I can create my reconnect handler or is it available in your library only?

demirschmid61, 2023-08-26:
Hi @matth-x 

Can you please share the send status notification example for 0.2.0 as well?

matth-x, 2023-08-28:
Can you take the `void authorize(...)` function from ArduinoOcpp.cpp as an example and rebuild it with StatusNotification? I'm sorry about not providing code snippets for version 0.2.0 anymore. Anyway, I believe that building an own `statusNotification` function would be the perfect tutorial for this library. Getting into the source code to some degree is actually very often actually.

demirschmid61, 2023-08-28:
Hi @matth-x,

Thank you very much, the problem is solved.

But when I am creating the configuration for metervalue in 0.3.0. Its a throwing error. 

  std::shared_ptr<MicroOcpp::Configuration<const char *>> mvSampledData = MicroOcpp::declareConfiguration<const char *>("MeterValuesSampledData", "DEFAULT VALUE", CONFIGURATION_FN);
  *mvSampledData = "Voltage,Current.Import,Energy.Active.Import.Register,Frequency, Power.Active.Import,Temperature"; // example: update configuration value
  MicroOcpp::configuration_save();

**Error Log**
namespace "MicroOcpp" has no member "declareConfiguration"

Do i need to include any extra header files?

demirschmid61, 2023-08-28:
Hi @matth-x 

Thank you very much, now I am able to send status notification for 0.2.0 as well as 0.3.0.

demirschmid61, 2024-02-07:
@matth-x 

I upgraded the library from ArduinoOcpp to MicroOcpp, facing an issue while resending status notifications after web socket disconnection.

**My code snippet is as follows.**

```
    for (size_t connectorId = 0; connectorId < numberOfConnector; connectorId++)
    {
        auto statusNotification = makeRequest(new StatusNotification(connectorId, 
                                                                           context->getModel().getConnector(connectorId)->getStatus(),
                                                                           context->getModel().getClock().now(), nullptr));

        statusNotification->setTimeout(60000);
        context->initiateRequest(std::move(statusNotification));
    }  
```

**The logs are**

[MO] info (StatusNotification.cpp:50): New status: Available (connectorId 0)
[MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): StopTransaction
[MO] info (StatusNotification.cpp:50): New status: Preparing (connectorId 1)
[MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): StatusNotification
[MO] info (StatusNotification.cpp:50): New status: Preparing (connectorId 2)
[MO] info (RequestQueueStorageStrategy.cpp:162): Replace cached operation (cache full): StatusNotification
[MO] Recv: WS pong

[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] Recv: WS pong

[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] Recv: WS pong

[MO] Recv: WS ping
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] info (RequestQueue.cpp:45): operation timeout: StatusNotification
[MO] Recv: WS pong

closed 2024-02-10 by demirschmid61