matth-x/MicroOcpp #181: Getting IdTag from RemoteStartTransaction()

shardul10, 2023-08-30:
Hi @matth-x 

I am still using 0.1.0 version. I haven't upgraded to 0.3.0 yet as I see several changes are required on my side. I will contact you separately for that. 

Hope you can help with this issue on 0.1.0.

Everything is working fine except for the RemoteStartTransaction() scenario. I have enabled the  callback  setOnRemoteStartTransactionSendConf().

As mentioned in the ArduinoOcpp.h,

oid setOnRemoteStartTransactionSendConf(OnSendConfListener onSendConf); //important, energize the power plug here and **capture the idTag**

The problem I am facing is the payload I receive in the callback is for the Accepted response sent by stack.

How do I extract the idTag from the RemoteStartTransaction so that I can pass that to the StartTransaction()?

matth-x, 2023-08-31:
Hi @shardul10, sorry for the confusion, this comment is indeed wrong. `onSendConf` only gets the success status of the RemoteStartTransaction operation. To fetch the idTag from inside the callback, use the function `getSessionIdTag()`. Alternatively, the "ReceiveReq" listener does get the original idTag from the server.

Please reopen this if you have follow-up questions.

closed 2023-08-31 by matth-x

shardul10, 2023-08-31:
Got it. Thanks.
For now I worked around it by using setConnectorPluggedSampler() to initiate automatic StartTransaction() from the stack. I will also try getSessionIdTag(). No need to reopen.