matth-x/MicroOcpp #183: How to use local authorization list

ziyadtarek7, 2023-09-02:
Hello,
I'm trying to authorize tags while offline and I found that local authorization list is what I need.
I first start the micro controller and connect to OCPP server then I send the list from the server and disconnect the microcontroller from the network. 
But when I call Authorize function it times out every time.
Am I missing something?

matth-x, 2023-09-03:
Hi @ziyadtarek7, yes the `Authorize` operation times out when there is no internet connection. To fetch idTags from the locally stored authorization lists, use this function:

https://github.com/matth-x/MicroOcpp/blob/5d429ba47654f58d204b62408db7ffda69549cd7/src/MicroOcpp/Model/Authorization/AuthorizationService.h#L31

You can find a usage example in the Connector class:

https://github.com/matth-x/MicroOcpp/blob/5d429ba47654f58d204b62408db7ffda69549cd7/src/MicroOcpp/Model/ConnectorBase/Connector.cpp#L529

If your goal is to authorize an idTag for a charging session, there is a ready-to-use function which fully handles authorization, including the offline scenario and could suit your case well. The `beginTransaction(...)` tries to send an Authorize operation, checks the local list and finally sends a StartTransaction if everything goes well. It's documented in MicroOcpp.h

Please reopen if you have follow-up questions

closed 2023-09-03 by matth-x

demirschmid61, 2024-08-14:
@ziyadtarek7 @matth-x @nelopauselli 

Thank you for the discussion on local authorization list.

I have one questions, what is the prerequisite to use offline transactions like mentioned in following:

1. Need to send the local list from server to the client
2. Any configuration key i need to set/reset to allow offline transaction from local authorization list?

Could you please acknowledge on this?