matth-x/MicroOcpp #184: Remotely start Transaction

Navjot0005, 2023-09-06:
Hi @matth-x @pedro-fuoco ,
I working on ArduinoOcpp library and facing issue with transaction. I connected with CMS everything is ok.
But when i give command from cms to start transaction remotely the status is changing from available to preparing but after some time it says timeout.

**Below is the code i am using** 
// matth-x/MicroOcpp
// Copyright Matthias Akstaller 2019 - 2023
// MIT License

#include <Arduino.h>
#if defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
ESP8266WiFiMulti WiFiMulti;
#elif defined(ESP32)
#include <WiFi.h>
#else
#error only ESP32 or ESP8266 supported at the moment
#endif

#include <ArduinoOcpp.h>
#include <ArduinoOcpp/Core/Configuration.h>

#define STASSID 
#define STAPSK 

/*charge HQ*/

/*steve instance*/

void setup() {

    /*
     * Initialize Serial and WiFi
     */ 

    Serial.begin(115200);

    Serial.print(F("[main] Wait for WiFi: "));

#if defined(ESP8266)
    WiFiMulti.addAP(STASSID, STAPSK);
    while (WiFiMulti.run() != WL_CONNECTED) {
        Serial.print('.');
        delay(1000);
    }
#elif defined(ESP32)
    WiFi.begin(STASSID, STAPSK);
    while (!WiFi.isConnected()) {
        Serial.print('.');
        delay(1000);
    }
#else
#error only ESP32 or ESP8266 supported at the moment
#endif

    Serial.println(F(" connected!"));

    /*
     * Initialize the OCPP library
     */
    ocpp_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL, "J1", "test");

    /*
     * Integrate OCPP functionality. You can leave out the following part if your EVSE doesn't need it.
     */
     setEnergyMeterInput([]() {
         //take the energy register of the main electricity meter and return the value in watt-hours
        
         return 10.f; 
     });
   
   // setEnergyMeterInput([](){int EnergyMeter;return EnergyMeter;}, 1);

//addMeterValueInput([]{return 120;}, "Voltage", "V", nullptr, nullptr);
addMeterValueInput([](){return 10;}, "Power.Active.Import", "kW", nullptr, nullptr, 1);
addMeterValueInput([](){return 100;}, "Energy.Active.Import.Register", "kWh", nullptr, nullptr, 1);

addMeterValueInput([](){return 120;}, "Voltage", "V", nullptr, "L1", 1);

addMeterValueInput([](){return 120;}, "Current.Import", "A", nullptr, "L1", 1);

std::shared_ptr<ArduinoOcpp::FilesystemAdapter> filesystem = ArduinoOcpp::makeDefaultFilesystemAdapter(ArduinoOcpp::FilesystemOpt::Use_Mount_FormatOnFail);
ArduinoOcpp::configuration_init(filesystem);

ArduinoOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "DEFAULT VALUE", CONFIGURATION_FN); //declare configuration and use the default value parameter (replace with required value)
ArduinoOcpp::configuration_save(); //again, not really necessary

  setSmartChargingCurrentOutput([](float limit) {
        //set the SAE J1772 Control Pilot value here
        Serial.printf("[main] Smart Charging allows maximum charge rate: %.0f\n", limit);
    });

    setConnectorPluggedInput([]() {
        //return true if an EV is plugged to this EVSE
        
        return false;
    },1);  
 }

void loop() {

    /*
     * Do all OCPP stuff (process WebSocket input, send recorded meter values to Central System, etc.)
     */
    ocpp_loop();
  
    /*
     * Energize EV plug if OCPP transaction is up and running
     */
    if (ocppPermitsCharge()) {
        //OCPP set up and transaction running. Energize the EV plug here
    } else {
        //No transaction running at the moment. De-energize EV plug
    }

    /*
     * Use NFC reader to start and stop transactions
     */
    if (/* RFID chip detected? */ false) {
        String idTag = "0123456789ABCD"; //e.g. idTag = RFID.readIdTag();

        if (!getTransaction()) {
            //no transaction running or preparing. Begin a new transaction
            Serial.printf("[main] Begin Transaction with idTag %s\n", idTag.c_str());

            /*
             * Begin Transaction. The OCPP lib will prepare transaction by checking the Authorization
             * and listen to the ConnectorPlugged Input. When the Authorization succeeds and an EV
             * is plugged, the OCPP lib will send the StartTransaction
             */
            auto ret = beginTransaction(idTag.c_str());

            if (ret) {
                Serial.println(F("[main] Transaction initiated. OCPP lib will send a StartTransaction when" \
                                 "ConnectorPlugged Input becomes true and if the Authorization succeeds"));
            } else {
                Serial.println(F("[main] No transaction initiated"));
            }

        } else {
            //Transaction already initiated. Check if to stop current Tx by RFID card
            if (idTag.equals(getTransactionIdTag())) {
                //card matches -> user can stop Tx
                Serial.println(F("[main] End transaction by RFID card"));

                endTransaction();
            } else {
                Serial.println(F("[main] Cannot end transaction by RFID card (different card?)"));
            }
        }
    }

   
}

**Below is the ESP32 side log**

[AO] info (ArduinoOcpp.cpp:376): Added ConnectorPluggedInput. Transaction-management is in auto mode now
[AO] info (Connection.cpp:56): Connected to url: /ocpp16/205894
[AO] info (BootNotification.cpp:83): request has been Accepted
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 0)
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 1)
[AO] info (SmartChargingService.cpp:523): New limit for connector 1, scheduled at = 2023-09-06T09:38:07.565Z, nextChange = 2037-01-01T00:00:00.000Z, limit = {-1.000000,-1.000000,-1}
[AO] warning (MeteringService.cpp:73): could not find metering device for Current.Offered
[AO] warning (ChangeConfiguration.cpp:153): validation failed for key=MeterValuesSampledData value=Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Voltage
[AO] info (TriggerMessage.cpp:29): Execute for message type StatusNotification, connectorId = -1
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 0)
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 1)
[AO] info (StatusNotification.cpp:49): New status: Preparing (connectorId 1)
[AO] info (Connector.cpp:193): Session mngt: timeout
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 1)

@matth-x Please guide me the path.

matth-x, 2023-09-06:
Hi @Navjot0005,

The ConnectorPluggedInput tells the OCPP library if the EV is plugged at the EVSE at the moment. In your code its value is always false so the library assumes that nothing is ever plugged. Remove `setConnectorPluggedInput` and try it again.

Please reopen if you have follow up questions.

closed 2023-09-06 by matth-x

Navjot0005, 2023-09-06:
Hi @matth-x  thanks for the reply.
Now I can start and stop transaction remotely thanks for the support.
Please can you tell me how this ocppPermitsCharge() will be true?
Because I want to turn the relays ON here.Pls suggest

if (ocppPermitsCharge()) {
        //OCPP set up and transaction running. Energize the EV plug here  
       
    }

Navjot0005, 2023-09-06:
@matth-x  Ok got it.
It will be true when charging starts.
Thanks very much @matth-x .