matth-x/MicroOcpp #185: setEnergyMeterInput([]() {}) and addMeterValueInput([](){return 10;}

Navjot0005, 2023-09-06:
Hi @matth-x 

Can you please tell me the difference between the **setEnergyMeterInput([]() {});**  and addMeterValueInput([](){return 10;}

/*
     * Integrate OCPP functionality. You can leave out the following part if your EVSE doesn't need it.
     */
    setEnergyMeterInput([]() {
        //take the energy register of the main electricity meter and return the value in watt-hours
        
         return 10.f;
     });

addMeterValueInput([](){return 10;}, "Power.Active.Import", "kW", nullptr, nullptr, 1);

Navjot0005, 2023-09-06:
Hi @matth-x 
After following this #110 
I am trying to send the energy meter value to the cms I am able to send the kwh and kw.
but i am not able to send voltage and current i followed #110 even then cms not getting the voltage and current value.
below is the code pls correct me.

/*Below in the logs you can see that there are two warnings as 
**[AO] warning (MeteringService.cpp:73): could not find metering device for Current.Import**
**[AO] warning (ChangeConfiguration.cpp:153): validation failed for key=MeterValuesSampledData value=Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Voltage** why I am getting these warning how it can find metering device where to putt this???*/
How I can correct this??

// matth-x/MicroOcpp
// Copyright Matthias Akstaller 2019 - 2023
// MIT License

#include <Arduino.h>
#if defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
ESP8266WiFiMulti WiFiMulti;
#elif defined(ESP32)
#include <WiFi.h>
#else
#error only ESP32 or ESP8266 supported at the moment
#endif

#include <ArduinoOcpp.h>
#include <ArduinoOcpp/Core/Configuration.h>

 bool connectorPlugged = false; //example for the variable where you keep the connector state
void setup() {

    /*
     * Initialize Serial and WiFi
     */ 

    Serial.begin(115200);

    Serial.print(F("[main] Wait for WiFi: "));

#if defined(ESP8266)
    WiFiMulti.addAP(STASSID, STAPSK);
    while (WiFiMulti.run() != WL_CONNECTED) {
        Serial.print('.');
        delay(1000);
    }
#elif defined(ESP32)
    WiFi.begin(STASSID, STAPSK);
    while (!WiFi.isConnected()) {
        Serial.print('.');
        delay(1000);
    }
#else
#error only ESP32 or ESP8266 supported at the moment
#endif

    Serial.println(F(" connected!"));

    /*
     * Initialize the OCPP library
     */
    ocpp_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL, "J1", "Tronic");

    /*
     * Integrate OCPP functionality. You can leave out the following part if your EVSE doesn't need it.
     */
     setEnergyMeterInput([]() {
         //take the energy register of the main electricity meter and return the value in watt-hours
        
         return 10.f;
[... 67 lines cut ...]
                endTransaction();
            } else {
                Serial.println(F("[main] Cannot end transaction by RFID card (different card?)"));
            }
        }
    }
}

**Pls find Logs below**

[main] Wait for WiFi: . connected!
[AO] info (Connection.cpp:56): Connected to url: /ocpp16/205894
[AO] info (BootNotification.cpp:83): request has been Accepted
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 0)
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 1)
[AO] info (SmartChargingService.cpp:523): New limit for connector 1, scheduled at = 2023-09-06T15:28:41.193Z, nextChange = 2037-01-01T00:00:00.000Z, limit = {-1.000000,-1.000000,-1}
**[AO] warning (MeteringService.cpp:73): could not find metering device for Current.Import**
**[AO] warning (ChangeConfiguration.cpp:153): validation failed for key=MeterValuesSampledData value=Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Voltage**
[AO] info (TriggerMessage.cpp:29): Execute for message type StatusNotification, connectorId = -1
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 0)
[AO] info (StatusNotification.cpp:49): New status: Available (connectorId 1)
[AO] info (Connector.cpp:224): Session mngt: trigger StartTransaction
[AO] info (StartTransaction.cpp:48): StartTransaction initiated
[AO] info (StatusNotification.cpp:49): New status: Charging (connectorId 1)
[AO] info (StartTransaction.cpp:142): Request has been accepted

chandan-ultraviolette, 2023-11-22:
For unable to send voltage and current, update the highlighted code in the MeteringService.cpp.

[image]

swordliongit, 2024-04-08:
Same issue. My addMeterValueInput functions don't add anything to the Meter Value request that is sent. Can't send Voltage and Current values. Any idea?