matth-x/MicroOcpp #187: Question about setOnDownload() vs setOnInstall()

shardul10, 2023-09-07:
@matth-x Based on the example in the code (built in FW service), it appears that you need to use only one of these, but not both. Can you please explain the difference? Since I am using MOCPP_CUSTOM_WS, I am getting the FirmwareService initialised in MicroOcpp.cpp using:

`FirmwareService *fwService = getOcppContext()->getModel().getFirmwareService();
`
then reusing rest of the implementation from example.

I understand the downloaded file will be copied to OTA partition by HTTPUpdate::handleUpdate(). Right?

matth-x, 2023-09-16:
Hi @shardul10,

The internal OTA setup is only for demonstration purposes (no TLS support or authenticity checks) and works only in combination with the Links2004 WebSocket library as you already pointed out. For your own integration you're free to use whatever OTA library you want. If I remember correctly, I already tried this OTA lib for secure updates over TLS and a FW signature check, and it worked pretty well. Nevertheless, the default integration does not use any of that, because it is inherently difficult to provide a good integration of that within another software product.

To integrate the real OTA functionality of your controller with OCPP, take the integration in the OpenEVSE project as a reference: https://github.com/OpenEVSE/openevse_esp32_firmware/blob/827b64591e6ea6786b45403f65edff621235c73b/src/ocpp.cpp#L597-L631

The principle is always the same: the OCPP lib receives an update URL and forwards it to your firmware in the onInstall callback. When this callback is executed, you should start OTA (using a library of your choice). During the update the OCPP lib regularly checks the status and the progress using the InstallationStatusInput.

Note that your controller should restart after the update was successful. The OCPP lib will send a success message to the server if the "firmwareVersion" passed to the `mocpp_initialize` ChargerCredentials is different.

Feel free to reopen this if you have follow-up questions.

closed 2023-09-16 by matth-x

shardul10, 2023-09-16:
I think I have implemented quire similar to the example you have given (have to test it). The only clarification I need is that after implementing everything in setOnInstall(), there is not really any need to implement setOnDownload(), right?

matth-x, 2023-09-16:
Yes, setOnDownload() is not needed.

shardul10, 2024-02-14:
I didn't understand the question. I tested it with Steve OCPP server. You have to download the firmware from the URL provided in the message in the OTA partition and restart the ESP32. It will boot with the new firmware. Hope that helps.

demirschmid61, 2024-02-15:
@shardul10, 
Thank you, Shardul, yesterday I tried and it worked in my case.

Can you please share your views on the #259 problem?