matth-x/MicroOcpp #188: Porting MicroOcpp(ArduinoOcpp) OCPP 1.6 Client for Quectel

chandan-ultraviolette, 2023-09-13:
"I want to use it with a Non-ESP based project, which utilizes a Quectel-based processor and SDK with FreeRTOS (all in C). I cloned the MicroOCPP repository and later discovered a dependency with ArduinoJSON, so I cloned it inside the MicroOCPP directory. The current directory structure is as follows:

MicroOCPP
ArduinoJSON
docs
src
..
..
CMakeLists.txt
I need to modify the MicroOCPP CMakeLists.txt file to fit this structure, but I'm encountering some compile-time issues. To address these issues, I rewrote the CMakeLists.txt file (attached). After doing so, the 'unable to find files' error was resolved, but I am now encountering strange errors related to Quectel SDK's pre-compiled GCC toolchain (logs attached: report.txt).

[image]

PS:
1. I've set the MOCPP_PLATFORM as MOCPP_PLATFORM_NONE.
2. So far, I haven't added the MicroOCPP loop call from the main. I plan to do this once I am able to compile successfully.
3. Regarding Mongoose, I prefer using the networking library part of Quectel SDK so didn't clone it. That part also I need to do after solving successful compilation. 

[CMakeLists.txt](https://github.com/matth-x/MicroOcpp/files/12594484/CMakeLists.txt)
[report.txt](https://github.com/matth-x/MicroOcpp/files/12594487/report.txt)

matth-x, 2023-09-13:
The error messages don't really seem to point into the MicroOcpp sources. To track this down I would take look at the build system first:

Can you start a new project (without MO or ArduinoJson) and compile it? Can you add a C++ file and link it with the binary? What happens if you include `<memory>` and `<ostream>` in the C++ file?

No problem to use the native networking library instead of Mongoose. You need to write an adapter which creates and maintains a WebSocket connection to the server, sends messages and forwards incoming messages to the OCPP library. The adapter will have to implement the `Connection` interface.

For an easier start on the new platform, I would recommend to postpone writing a custom Connection and using the internal Loopback class instead:

https://github.com/matth-x/MicroOcpp/blob/8e3995b12fd22dcffa62b0ce3e5b6cd87c966634/tests/ocppEngineLifecycle.cpp#L11-L13

chandan-ultraviolette, 2023-09-13:
I was facing a similar issue when created a new project (without MO or ArduinoJson) and added a C++ file. Changes need to be made in CMakeLists.txt. I will connect with Quectel SDK folks on this.

As this may take some time meanwhile anything else can I try ?

chandan-ultraviolette, 2023-10-17:
Also I am looking to know how to test it without any paid CMS server.

closed 2023-10-17 by chandan-ultraviolette