matth-x/MicroOcpp #191: Issue with implementing reservation

ziyadtarek7, 2023-09-16:
Hello,

I'm trying to handle reservations in my project. I send reservation request from the server and handle it properly in my esp32 code but I get and error during runtime that causes the esp to panic and reboot.

Here is the part of code that causes the error:
 `auto con = getOcppContext();`
 `auto res = con->getModel().getReservationService()->getReservation(1);`
 `reserved = res->isActive(); `
 
 the moment isActive() is called I get this error on my serial monitor:
 
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Am I doing it wrong?

Thank you

matth-x, 2023-09-16:
`getReservation()` returns nullptr if there is no reservation at the moment. For some reason, the OCPP lib is not aware of any reservation at that moment, but it's a little hard to guess why the reservation failed with only that information. Can you maybe capture the original ReserveNow message and post it here?

Since `getReservation()` is not guaranteed to return a valid pointer, always check the result for null:

```cpp
auto res = con->getModel().getReservationService()->getReservation(1);
if (res) {
    //now it is safe to access res
    reserved = res->isActive();
}
```

shardul10, 2023-09-17:
@matth-x will the OCPP library check if the reservation exists and take appropriate action (e.g. accept or reject when responding to RemoteStartTransaction()) or the firmware will have to check for reservation, compare idTags and accept/reject RemoteStartTransaction() explicitly? Based on this thread, I assume the firmware has to do this. Can you please confirm?

matth-x, 2023-09-17:
Reservations fall under the built-in transaction management, so if using the high-level functions for transactions in MicroOcpp.h (`beginTransaction(...)` and `endTransaction(...)`), no action is required. In contrast, the low-level functions `startTransaction(...)` and `stopTransaction(...)` always send the corresponding operation and if using these, a reservation check is required before initiating the operation. Also see the documentation of `beginTransaction(...)`:

https://github.com/matth-x/MicroOcpp/blob/7a72b6bc0579ccf0282160813c7a051a22f18841/src/MicroOcpp.h#L112-L128

The library also rejects RemoteStartTransaction requests if the connector is blocked by a reservation.

Nevertheless, in many scenarios it is interesting to fetch the current reservation status, for example if the charger has a display. Then the method shown by @ziyadtarek7 is the best practice.

shardul10, 2023-09-17:
Thanks. I think then I might have run into an issue where the reservation is not honoured by the library. I am relying on library to issue StartTransaciton() using setConnectorPluggedInput(). See the message exchange below. Reservation accepted for one idTag, but RemoteTransaction.conf is sent for another idTag followed by StartTransaction.

00000059521 INF [operator()] [2,"c4945b46-f066-48ef-8b6f-1e3a08b39206","ReserveNow",{"connectorId":1,"expiryDate":"2023-09-17T18:00:00.000Z","idTag":"**9209102911**","reservationId":6}]
00000059772 INF [sendTXT] [3,"c4945b46-f066-48ef-8b6f-1e3a08b39206",{"status":"Accepted"}]
00000059805 INF [sendTXT] [2,"1000003","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Reserved","timestamp":"2023-09-17T11:09:29.524Z"}]
00000060276 INF [operator()] [3,"1000003",{}]
00000089627 INF [operator()] [2,"0219ce76-f479-4e12-9458-9ff613b07ca5","RemoteStartTransaction",{"connectorId":1,"idTag":"**8203910913**"}]
00000089785 INF [sendTXT] [3,"0219ce76-f479-4e12-9458-9ff613b07ca5",{"status":"Accepted"}]
00000090239 INF [sendTXT] [2,"1000004","StartTransaction",{"connectorId":1,"idTag":"**8203910913**","meterStart":0,"timestamp":"2023-09-17T11:09:59.539Z"}]
00000090547 INF [operator()] [3,"1000004",{"transactionId":1,"idTagInfo":{"status":"Accepted","expiryDate":"2023-09-17T12:10:01.271Z"}}]
00000090899 INF [sendTXT] [2,"1000005","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2023-09-17T11:10:00.004Z"}]
00000091264 INF [operator()] [3,"1000005",{}]

```

ziyadtarek7, 2023-09-17:
> `getReservation()` returns nullptr if there is no reservation at the moment. For some reason, the OCPP lib is not aware of any reservation at that moment, but it's a little hard to guess why the reservation failed with only that information. Can you maybe capture the original ReserveNow message and post it here?
> 
> Since `getReservation()` is not guaranteed to return a valid pointer, always check the result for null:
> 
> ```c++
> auto res = con->getModel().getReservationService()->getReservation(1);
> if (res) {
>     //now it is safe to access res
>     reserved = res->isActive();
> }
> ```

Here is the reserve now message:
```
[INFO ] 2023-09-17 15:29:41,459 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (SteVe-Executor-0) - [chargeBoxId=EVFlash_1, sessionId=a7b21d88-3c8f-ceb4-ae62-84d9da9fcb0c] Sending: [2,"801c0095-712a-42b1-b25a-5417990ce629","ReserveNow",{"connectorId":1,"expiryDate":"2023-09-18T06:00:00.000Z","idTag":"test_tag","reservationId":6}]
```

The original error was solved but now another one arouse.
If a connector is reserved for an specific idTag and another idTag tried to start transaction, it will be accepted and charging starts.
so does beginTransaction() function check for reservation or should I check manually?

Here are server side logs for this:

```
[INFO ] 2023-09-17 15:50:09,526 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (SteVe-Executor-0) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Sending: [2,"d85f0b38-de85-4b44-8ab5-930edf6854ac","ReserveNow",{"connectorId":1,"expiryDate":"2023-09-18T00:00:00.000Z","idTag":"51.246.172.16","reservationId":10}]
[INFO ] 2023-09-17 15:50:10,408 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1209770703-49) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Received: [3,"d85f0b38-de85-4b44-8ab5-930edf6854ac",{"status":"Accepted"}]
[INFO ] 2023-09-17 15:50:11,414 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1209770703-38) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Received: [2,"1000005","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Reserved","timestamp":"2023-09-17T15:50:10.353Z"}]
[INFO ] 2023-09-17 15:50:11,420 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1209770703-38) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Sending: [3,"1000005",{}]
[INFO ] 2023-09-17 15:50:47,739 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1209770703-49) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Received: [2,"1000006","Authorize",{"idTag":"163.64.180.158"}]
[INFO ] 2023-09-17 15:50:47,744 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1209770703-49) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Sending: [3,"1000006",{"idTagInfo":{"status":"Accepted","expiryDate":"2023-09-17T16:50:47.744Z"}}]
[INFO ] 2023-09-17 15:50:50,162 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1209770703-47) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Received: [2,"1000007","StartTransaction",{"connectorId":1,"idTag":"163.64.180.158","meterStart":0,"timestamp":"2023-09-17T15:50:48.980Z"}]
[INFO ] 2023-09-17 15:50:50,179 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1209770703-47) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Sending: [3,"1000007",{"transactionId":9,"idTagInfo":{"status":"Accepted","expiryDate":"2023-09-17T16:50:50.165Z"}}]
[INFO ] 2023-09-17 15:50:51,288 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp1209770703-49) - [chargeBoxId=EVFlash_1, sessionId=22f8ad7e-76bc-afd8-dcb1-08e3bccf1e78] Received: [2,"1000008","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2023-09-17T15:50:50.107Z"}]
```

matth-x, 2023-09-17:
@shardul10 Seems like I was wrong here. In my implementation the RemoteStartTransaction does NOT check for reservations. However, I'm not so sure if that's required at all. The OCPP specification doesn't describe this case and I would also tend to say that if the backend sends a RemoteStartTransaction with another idTag than the reservation, it probably has a good reason to overrule the previous reservation. I will keep an eye on this case and discuss it with different backend operators though. Maybe I will eventually change it.

@ziyadtarek7 Yes, beginTransaction checks for reservations. Did you use that function when capturing these logs?

ziyadtarek7, 2023-09-17:
> @shardul10 Seems like I was wrong here. In my implementation the RemoteStartTransaction does NOT check for reservations. However, I'm not so sure if that's required at all. The OCPP specification [...]

@matth-x 
Yes, I used beginTransaction, and I also checked for the reservation using the getReservation API and I was able to get idtag for that reservation. So I'm not sure what's happening here

matth-x, 2023-09-17:
Okay, let me take a deeper look into this, could also be a defect. Can you share the result of the GetConfiguration?

ziyadtarek7, 2023-09-18:
> Okay, let me take a deeper look into this, could also be a defect. Can you share the result of the GetConfiguration?

Here is the response for Get Configuration request:
[image]

matth-x, 2023-09-20:
@ziyadtarek7 Thanks for providing further information. It could mean that the regression tests failed for Reservation, I will check that very soon. In any case the `beginTransaction(...)` is specified to check for reservations so you can assume that for your development at the moment and get the fixed version later.

matth-x, 2023-09-24:
@ziyadtarek7 Thanks again for reporting this. Bug reports like this are incredibly helpful for an open source project.

Now, beginTransaction should behave as expected again on the develop branch.

PR #196 (2023-09-24): Unit tests for Reservation
Unit tests for the Reservation feature profile, corresponding fixes and minor cosmetic changes to simplify the handling of the ReservationService in the test cases.

Fixes #191 and #154.
closed 2023-09-24 by matth-x