matth-x/MicroOcpp #193: Saving offline metervalues

VjSargunam, 2023-09-21:
Hi Matt,
            I want to save the offline meter value in the local storage (say in my case in SD card) and update the CMS with this saved meter value data once connection is reestablished between CMS and charger. How to store the offline meter values in local storage? Help me in this regard.

Thanks
VJsargunam

matth-x, 2023-09-23:
Hi @VjSargunam,

The long-time storage of the MeterValues operation itself is not supported at the moment. But in case the measurements are used for billing purposes, there is a very similar use case which is supported. The StopTransaction embeds a field "transactionData" which contains a history of metering data which is similar to the MeterValues. Here are the steps to activate it:

- Change the configurations `StopTxnSampledData` or `StopTxnAlignedData` so that they contain the desired measurands
- Change `Cst_StopTxnDataCapturePeriodic` (the `Cst_` prefix could be different depending on your version) to `true` so that the library will not only store the measurands at the begin and end of the transaction, but also periodically in between
- At the moment, the size of the history is limited to only 4 snapshots. To increase it, you need to modify [MOCPP_MAX_STOPTXDATA_LEN](https://github.com/matth-x/MicroOcpp/blob/7a72b6bc0579ccf0282160813c7a051a22f18841/src/MicroOcpp/Model/Metering/MeterStore.cpp#L12) by hand (will be variable in future)

The primary use case for this is to store signed [OCMF](https://github.com/SAFE-eV/OCMF-Open-Charge-Metering-Format) data sets which are necessary for public chargers in Germany. But generally it's suitable for any measurement format.

Will it work for you? I can also outline how to extend the operations store so that it also supports MeterValues, but I don't have any plans to add this feature to this repo in near future.

VjSargunam, 2023-09-26:
Hi Matt,
            As per your suggestion i changed the StopTxnDataCapturePeriodic as shown below
StopTxnDataCapturePeriodic = declareConfiguration<bool>(MOCPP_CONFIG_EXT_PREFIX "StopTxnDataCapturePeriodic", true, CONFIGURATION_FN, true, true, true, false);

And after doing remote start and remote stop the transaction how to visualize the saved meter data?. Where to get this stored measurand data?

matth-x, 2023-09-26:
The snapshots are stored as files in the OCPP directory. Their naming scheme is `sd-X-Y-Z.jsn` whereas X is the connectorId, Y is the transaction number (a local transaction key, not the same as the transactionId) and Z is an incrementing number for the snapshot.

They should also be transmitted in the StopTransaction.req if everything works, so you can check the results in the logs.

VjSargunam, 2023-09-27:
Hi Matt,
              As you suggested i changed the *StopTxnDataCapturePeriodic  value to true. But once i compiled and burn the code in to esp32, the esp32 gets crashed. 
I changed the code present in MicroOcpp/Model/Metering/MeteringConnector.cpp like below
`//StopTxnDataCapturePeriodic= declareConfiguration<bool>(MOCPP_CONFIG_EXT_PREFIX "StopTxnDataCapturePeriodic", false, CONFIGURATION_FN, true, true, true, false);
*StopTxnDataCapturePeriodic = true;`
the output is like
`[main] Wait for WiFi: ... connected!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dbbce  PS      : 0x00060530  A0      : 0x800fb854  A1      : 0x3ffb24a0  
A2      : 0x00000000  A3      : 0x3ffb2578  A4      : 0x3ffc350c  A5      : 0x3ffcf780  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x800d9e0a  A9      : 0x3ffb2480  
A10     : 0x3ffcaf9c  A11     : 0x3ffb24d8  A12     : 0x00000001  A13     : 0x3ffcf774  
A14     : 0x00000000  A15     : 0x3ffcf780  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000022  LBEG    : 0x4008a48d  LEND    : 0x4008a49d  LCOUNT  : 0xfffffffe  

Backtrace:0x400dbbcb:0x3ffb24a00x400fb851:0x3ffb2500 0x400ec313:0x3ffb25a0 0x400d7b9b:0x3ffb2630 0x400d7ed1:0x3ffb2660 0x400d31a1:0x3ffb27b0 0x400fff6e:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...`

And also, if I want to change the configuration from OCPP server what are all the key values i have to set for StopTxnSampledData or StopTxnAlignedData

matth-x, 2023-09-27:
You still need to declare the configuration before editing it. In the code snippet above, the declaration is commented out and thereby all subsequent writes and reads will be redirected to the NULL pointer and crash. If you revert the code changes and use the OCPP server to send a ChangeConfiguration to enable Cst_StopTxnDataCapturePeriodic, it might be a bit clearer.

Anyway, the StopTx looks already good, except that it only contains the start and stop measurement for now. On the flash storage, there should be two files `sd-*.jsn` now containing the record.

The OCPP lib rejects unsupported measurands in the selections (e.g. StopTxnSampledData). To add support for more measurands, see this example:

https://github.com/OpenEVSE/openevse_esp32_firmware/blob/8c9a482c912005f4c5f0e2d82d2f596e665dd6eb/src/ocpp.cpp#L206-210

VjSargunam, 2023-09-27:
Hi Matt,
               I made the changes as you mentioned like below
`StopTxnDataCapturePeriodic= declareConfiguration<bool>(MOCPP_CONFIG_EXT_PREFIX "StopTxnDataCapturePeriodic", false, CONFIGURATION_FN, true, true, true, false);
    *StopTxnDataCapturePeriodic = true;`
but i didn't see any stored meter values in the log
this is the log i got
`StatusNotification	[2,"1000020","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2023-09-27T10:34:04.452Z"}]	[3,"1000020",{}]	9/27/2023 4:04:04 PM
StopTransaction	[2,"1000019","StopTransaction",{"meterStop":0,"timestamp":"2023-09-27T10:34:04.014Z","transactionId":409061,"reason":"Remote"}]	[3,"1000019",{"idTagInfo":{"expiryDate":"2023-09-27T10:34:04.507Z","parentIdTag":"","status":"Accepted"}}]	9/27/2023 4:04:04 PM
RemoteStopTransaction	[2,"310697e5-5ec3-4782-b9f3-09082e3c127a","RemoteStopTransaction",{"transactionId":409061}]	[3,"310697e5-5ec3-4782-b9f3-09082e3c127a",{"status":"Accepted"}]	9/27/2023 4:04:03 PM
Heartbeat	[2,"1000018","Heartbeat",{}]	[3,"1000018",{"currentTime":"2023-09-27T10:33:41.221Z"}]	9/27/2023 4:03:41 PM
MeterValues	[2,"1000017","MeterValues",{"connectorId":1,"transactionId":409061,"meterValue":[{"timestamp":"2023-09-27T10:33:12.817Z","sampledValue":[{"value":" 0.0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":" 0.0","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"230.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L1-N","location":"Outlet","unit":"V"},{"value":"232.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L2-N","location":"Outlet","unit":"V"},{"value":"229.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L3-N","location":"Outlet","unit":"V"},{"value":"16.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","location":"Outlet","unit":"A"},{"value":"15.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","location":"Outlet","unit":"A"},{"value":"17.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","location":"Outlet","unit":"A"},{"value":"37.0","context":"Sample.Periodic","measurand":"Temperature","location":"Cable","unit":"Celsius"},{"value":"25.0","context":"Sample.Periodic","measurand":"SoC","location":"EV","unit":"Percent"}]}]}]	[3,"1000017",{}]	9/27/2023 4:03:12 PM
MeterValues	[2,"1000016","MeterValues",{"connectorId":1,"transactionId":409061,"meterValue":[{"timestamp":"2023-09-27T10:32:12.817Z","sampledValue":[{"value":" 0.0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":" 0.0","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"230.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L1-N","location":"Outlet","unit":"V"},{"value":"232.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L2-N","location":"Outlet","unit":"V"},{"value":"229.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L3-N","location":"Outlet","unit":"V"},{"value":"16.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","location":"Outlet","unit":"A"},{"value":"15.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","location":"Outlet","unit":"A"},{"value":"17.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","location":"Outlet","unit":"A"},{"value":"37.0","context":"Sample.Periodic","measurand":"Temperature","location":"Cable","unit":"Celsius"},{"value":"25.0","context":"Sample.Periodic","measurand":"SoC","location":"EV","unit":"Percent"}]}]}]	[3,"1000016",{}]	9/27/2023 4:02:12 PM
Heartbeat	[2,"1000015","Heartbeat",{}]	[3,"1000015",{"currentTime":"2023-09-27T10:31:41.218Z"}]	9/27/2023 4:01:41 PM
MeterValues	[2,"1000014","MeterValues",{"connectorId":1,"transactionId":409061,"meterValue":[{"timestamp":"2023-09-27T10:31:12.815Z","sampledValue":[{"value":" 0.0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":" 0.0","context":"Sample.Periodic","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"230.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L1-N","location":"Outlet","unit":"V"},{"value":"232.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L2-N","location":"Outlet","unit":"V"},{"value":"229.0","context":"Sample.Periodic","measurand":"Voltage","phase":"L3-N","location":"Outlet","unit":"V"},{"value":"16.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","location":"Outlet","unit":"A"},{"value":"15.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L2","location":"Outlet","unit":"A"},{"value":"17.0","context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","location":"Outlet","unit":"A"},{"value":"37.0","context":"Sample.Periodic","measurand":"Temperature","location":"Cable","unit":"Celsius"},{"value":"25.0","context":"Sample.Periodic","measurand":"SoC","location":"EV","unit":"Percent"}]}]}]	[3,"1000014",{}]	9/27/2023 4:01:12 PM
MeterValues	[2,"1000013","MeterValues",{"connectorId":1,"transactionId":409061,"meterValue":[{"timestamp":"2023-09-27T10:30:12.815Z","sampledValue":[{"value":" 0.0","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"Wh"},{"value":" 0.0","c [... cut ...] us":"Accepted"},"transactionId":409061}]	9/27/2023 3:57:12 PM
RemoteStartTransaction	[2,"61f25a7a-26e4-44dd-b039-982b41ad77e3","RemoteStartTransaction",{"connectorId":1,"idTag":"5900D490031E"}]	[3,"61f25a7a-26e4-44dd-b039-982b41ad77e3",{"status":"Accepted"}]	9/27/2023 3:57:12 PM
Heartbeat	[2,"1000006","Heartbeat",{}]	[3,"1000006",{"currentTime":"2023-09-27T10:25:41.218Z"}]	9/27/2023 3:55:41 PM`

how to enable the StopTXnDataCapturPeriodic from OCPP server since there is no configuration like it.

matth-x, 2023-09-28:
Thank you for the feedback. Feel free to reopen if you have follow-up questions.

closed 2023-09-28 by matth-x

VjSargunam, 2023-09-29:
Hi Matt,
             As per your instruction i added this peace of code in the setup function in order to add more key values to  StopTxnSampledData and StopTxnAlignedData but i can't able to add more keyvalues. 
MicroOcpp::configuration_init(filesystem);
    `MicroOcpp::declareConfiguration<const char*>(
        "MeterValuesSampledData", "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import,SoC,Temperature");
    MicroOcpp::declareConfiguration<bool>(
        "AO_PreBootTransactions", true);   
    /*
     * Initialize the OCPP library
     */
    mocpp_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL, "My Charging Station", "My company name");`