matth-x/MicroOcpp #194: How to add new metervalues

vanhcm20, 2023-09-22:
Yah, i know have 

> addMeterValueInput

and i founded
```
std::shared_ptr<ArduinoOcpp::Configuration<const char *>> mvSampledData = ArduinoOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "", CONFIGURATION_FN);
*mvSampledData = "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import";
ArduinoOcpp::configuration_save();
```

but its in old version, how i change for new MicroOcpp,
when i try replace ArduioOcpp by MicroOcpp
[image]

I use Arduino IDE
Finally, i think should have something like "Full example",

vanhcm20, 2023-09-23:
OK, i got it. just include path to code
`#include <MicroOcpp/Core/Configuration.h>`

but i cant see something update to my ocpp server
I use SteVe
I waiting more than 5 minutes and i just got
[image]

here my code (in setup())
```
mocpp_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL, CHARGER_ID, VENDORS);

    setSmartChargingPowerOutput([](float limit) {
        //set the SAE J1772 Control Pilot value here
        Serial.printf("[main] Smart Charging allows maximum charge rate: %.0f\n", limit);
    });

    setConnectorPluggedInput([]() {
        //return true if an EV is plugged to this (= J1772 State C)
        return true;
    });

    // setEvReadyInput([]() {
    //     //return true if EV is ready to charge (= PWM signal on)
    //     return true;
    // });

    // setEvseReadyInput([]() {
    //     //return true if EVSE is ready to charge
    //     return true;
    // }); 

    addMeterValueInput([](){return 1;}, "Power.Active.Import", "kW", nullptr, nullptr, 1);
    addMeterValueInput([](){return 2;}, "Energy.Active.Import.Register", "kWh", nullptr, nullptr, 1);

    addMeterValueInput([]()
                       { return 232; },
                       "Voltage", "V", nullptr, "L1", 1);
    addMeterValueInput([]()
                       { return 234; },
                       "Voltage", "V", nullptr, "L2", 1);
    addMeterValueInput([]()
                       { return 230; },
                       "Voltage", "V", nullptr, "L3", 1);

    addMeterValueInput([]()
                       { return 11; },
                       "Current.Import", "A", nullptr, "L1", 1);
    addMeterValueInput([]()
                       { return 12; },
                       "Current.Import", "A", nullptr, "L2", 1);
    addMeterValueInput([]()
                       { return 13; },
                       "Current.Import", "A", nullptr, "L3", 1);

    std::shared_ptr<MicroOcpp::Configuration<const char *>> mvSampledData = MicroOcpp::declareConfiguration<const char *>(
            "MeterValuesSampledData", "", CONFIGURATION_FN);
    *mvSampledData = "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import";
    MicroOcpp::configuration_save();
```

vanhcm20, 2023-09-23:
Haha, ok that my bad, i so tired in night.
i forget upload new code, i just complie it and wait.

[this](https://github.com/matth-x/MicroOcpp/issues/194#issuecomment-1732244329url) working

A few advice for people who come here are very important order, you have to Init before Addmetervalue, and finally Saveconfig

Sorry @matth-x cuz my disturb

closed 2023-09-23 by vanhcm20

demirschmid61, 2023-12-31:
@vanhcm20 

I am trying to update the OCPP library, but I am stuck in adding other meterValues.

I did the same thing in ArduinoOcpp and it is working, i am trying to replicate the same thing in microOcpp, I got following error.

/*************************************************************************/
src/main.cpp:2736:19: error: 'MicroOcpp::ConfigurationContainer' is not a template
   std::shared_ptr<MicroOcpp::ConfigurationContainer <const char *>> mvSampledData = MicroOcpp::declareConfiguration<const char *>("MeterValuesSampledData", "", CONFIGURATION_FN);
                   ^~~~~~~~~
src/main.cpp:2736:130: error: conversion from 'shared_ptr<MicroOcpp::Configuration>' to non-scalar type 'shared_ptr<MicroOcpp::ConfigurationContainer>' requested 
   std::shared_ptr<MicroOcpp::ConfigurationContainer <const char *>> mvSampledData = MicroOcpp::declareConfiguration<const char *>("MeterValuesSampledData", "", CONFIGURATION_FN);
                                                                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:2738:20: error: no match for 'operator=' (operand types are 'std::__shared_ptr_access<MicroOcpp::ConfigurationContainer, (__gnu_cxx::_Lock_policy)2, 
false, false>::element_type' {aka 'MicroOcpp::ConfigurationContainer'} and 'const char [53]')
   *mvSampledData = "Voltage,Current.Import,Energy.Active.Import.Register"; // example: update configuration value
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32-development-board/MicroOcpp/src/MicroOcpp/Core/Configuration.h:9,
                 from src/main.cpp:323:
.pio/libdeps/esp32-development-board/MicroOcpp/src/MicroOcpp/Core/ConfigurationContainer.h:15:7: note: candidate: 'MicroOcpp::ConfigurationContainer& MicroOcpp::ConfigurationContainer::operator=(const MicroOcpp::ConfigurationContainer&)'
 class ConfigurationContainer {
       ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-development-board/MicroOcpp/src/MicroOcpp/Core/ConfigurationContainer.h:15:7: note:   no known conversion for argument 1 from 'const char [53]' to 'const MicroOcpp::ConfigurationContainer&'
/***************************************************************************************************/

I have used the following code snippet to add meterValues

`  std::shared_ptr<MicroOcpp::Configuration <const char *>> mvSampledData = MicroOcpp::declareConfiguration<const char *>("MeterValuesSampledData", "", CONFIGURATION_FN);
  *mvSampledData = "Voltage,Current.Import,Energy.Active.Import.Register"; 
  MicroOcpp::configuration_save(); `

Also, included the following  header file

#include <MicroOcpp/Core/Configuration.h>

Am I missing anything??

Pablo2048, 2023-12-31:
Adding meter values are not configuration. I'm using it like this:
```cpp
    addMeterValueInput([=]() { return this->getL1Current(); }, "Current.Import", "A", nullptr, "L1"); // Current L1
    addMeterValueInput([=]() { return this->getL1Voltage(); }, "Voltage", "V", nullptr, "L1"); // Voltage L1
    addMeterValueInput([=]() { return this->getL2Voltage(); }, "Voltage", "V", nullptr, "L2"); // Voltage L2
    addMeterValueInput([=]() { return this->getL3Voltage(); }, "Voltage", "V", nullptr, "L3"); // Voltage L3
    addMeterValueInput([=]() { return this->getMaxCurrent(); }, "Current.Offered", "A", nullptr,
                       "L1"); // Max current, offered to vehicle
```

demirschmid61, 2024-01-01:
@Pablo2048, @vanhcm20

I also added the addMeterValueInput() function as well, but only energy is sending in meterValues payload. The logs are mentioned below, the charging cycle is perfectly working fine.

/***********************************Logs Started*********************************/
[MO] Send: [2,"1000000","BootNotification",{"chargePointModel":"My Charging Station","chargePointVendor":"My company name"}]
[MO] Recv: WS pong
[MO] Recv: [3,"1000000",{"currentTime":"2024-01-01T07:10:07.780303200Z","interval":900,"status":"Accepted"}]
[MO] info (BootNotification.cpp:83): request has been Accepted
[MO] info (StatusNotification.cpp:50): New status: Available (connectorId 0)
[MO] info (StatusNotification.cpp:50): New status: Preparing (connectorId 1)
[MO] Send: [2,"1000001","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-01-01T07:10:07.791Z"}]
[MO] Recv: [3,"1000001",{}]
[MO] Send: [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2024-01-01T07:10:07.793Z"}]
[MO] Recv: [3,"1000002",{}]
[MO] Recv: [2,"3c5dbfbb-1ab9-4454-8cc4-c6544255bb8c","RemoteStartTransaction",{"connectorId":1,"idTag":"amo9763383219"}]
[MO] Send: [3,"3c5dbfbb-1ab9-4454-8cc4-c6544255bb8c",{"status":"Accepted"}]
[MO] info (Connector.cpp:240): Session mngt: trigger StartTransaction
[MO] info (StartTransaction.cpp:48): StartTransaction initiated
[MO] Send: [2,"1000003","StartTransaction",{"connectorId":1,"idTag":"123456789","meterStart":0,"timestamp":"2024-01-01T07:10:19.444Z"}]
[MO] info (StatusNotification.cpp:50): New status: Charging (connectorId 1)
[MO] Recv: [3,"1000003",{"idTagInfo":{"status":"Accepted"},"transactionId":227284}]
[MO] info (StartTransaction.cpp:142): Request has been accepted
[MO] Send: [2,"1000004","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2024-01-01T07:10:19.686Z"}]
[MO] Recv: [3,"1000004",{}]
[MO] Recv: WS pong
[MO] Recv: WS pong
[MO] Recv: WS ping
[MO] Recv: WS pong
[MO] Recv: WS pong
**[MO] Send: [2,"1000005","MeterValues",{"connectorId":1,"transactionId":227284,"meterValue":[{"timestamp":"2024-01-01T07:11:19.669Z","sampledValue":[{"value":" 0.6","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"}]}]}]
[MO] Recv: [3,"1000005",{}]**
[MO] Recv: WS pong
[MO] Recv: WS pong
[MO] Recv: WS ping
[MO] Recv: WS pong
[MO] Recv: WS pong
[MO] Send: [2,"1000006","MeterValues",{"connectorId":1,"transactionId":227284,"meterValue":[{"timestamp":"2024-01-01T07:12:19.669Z","sampledValue":[{"value":" 0.9","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"}]}]}]
[MO] Recv: [3,"1000006",{}]
[MO] Recv: WS pong
[MO] Recv: [2,"473d837b-3075-412c-9d7f-e18141a24c04","RemoteStopTransaction",{"transactionId":227284}]
[MO] Send: [3,"473d837b-3075-412c-9d7f-e18141a24c04",{"status":"Accepted"}]
[MO] info (Connector.cpp:293): Session mngt: trigger StopTransaction
[MO] info (StopTransaction.cpp:50): StopTransaction initiated
[MO] Send: [2,"1000007","StopTransaction",{"meterStop":1,"timestamp":"2024-01-01T07:12:30.048Z","transactionId":227284,"reason":"Remote"}]
[MO] info (StatusNotification.cpp:50): New status: Finishing (connectorId 1)
[MO] Recv: [3,"1000007",{"idTagInfo":{"status":"Accepted"}}]
[MO] info (StopTransaction.cpp:197): Request has been accepted!
[MO] Send: [2,"1000008","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Finishing","timestamp":"2024-01-01T07:12:30.502Z"}]
[MO] Recv: [3,"1000008",{}]
[MO] Recv: WS pong
/***********************************Logs ends*********************************/

My setup function is also mentioned below, there i have added the **addMeterValueInput()** for energy, current and voltage as well.

```
float whImport = 0.00;

float readEnergy()
{
  whImport += 0.3;
  return whImport;
}
void setup()
{

  Serial.begin(115200);

  Serial.print(F("[main] Wait for WiFi: "));

  WiFi.begin(STASSID, STAPSK);
  while (!WiFi.isConnected())
  {
    Serial.print('.');
    delay(1000);
  }
  Serial.println(F(" connected!"));
  mocpp_initialize(OCPP_BACKEND_URL, OCPP_CHARGE_BOX_ID, "My Charging Station", "My company name");
  addMeterValueInput([=]
                     { return readEnergy(); },
                     "Energy.Active.Import.Register", "Wh", nullptr, nullptr); //Energy

  addMeterValueInput([=]()
                     { return 12.12; },
                     "Current.Import", "A", nullptr, "L1"); // Current L1
  addMeterValueInput([=]()
                     { return 222.22; },
                     "Voltage", "V", nullptr, "L1"); // Voltage L1
  pinMode(2, OUTPUT);
  // For simulation, added connector is plugged in
  setConnectorPluggedInput([]()
                           { return true; });
  endTransaction(nullptr, "Reboot", 1);
```

Can you please suggest me to get all the meterValues?

vanhcm20, 2024-01-02:
Hi @demirschmid61 
that how i addMeter, you need add 
```
    std::shared_ptr<MicroOcpp::Configuration<const char *>> mvSampledData = MicroOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "", CONFIGURATION_FN);
    *mvSampledData = "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import";
    // MicroOcpp::configuration_save(); //it really not important

```
after addMeter

its will look like this
```
    addMeterValueInput([]()
                       { return adc_value[cs_l3]; },
                       "Current.Import", "A", nullptr, "L3", 1);

    std::shared_ptr<MicroOcpp::Configuration<const char *>> mvSampledData = MicroOcpp::declareConfiguration<const char *>(
        "MeterValuesSampledData", "", CONFIGURATION_FN);
    *mvSampledData = "Power.Active.Import,Energy.Active.Import.Register,Voltage,Current.Import";
    // MicroOcpp::configuration_save(); //it really not important
```
Hoping its help you