matth-x/MicroOcpp #198: Port for NXP iMX RT1064

GanesanGuru, 2023-09-28:
Hi @team,

Is there any port/guidance available for implementing OCPP 1.6J feature to iMX RT1064 EVK board ? Quite important guys...

matth-x, 2023-09-28:
Hi @GanesanGuru,

Here is a general guide on how to integrate this library into custom platforms: https://github.com/matth-x/MicroOcpp/issues/172#issuecomment-1669574485

I will soon update that guide with the newest naming conventions. In the meantime, please also see these migration notes: https://matth-x.github.io/MicroOcpp/migration/

Feel free to reopen if you have follow-up questions.

closed 2023-09-28 by matth-x

GanesanGuru, 2023-11-09:
Hi @matth-x ,
With reference to your #172 commit,  for iMXRT1064-EVK with MCUXprsso IDE, I've done the following. Kindly, let me know whether I'm on right track...

1. Created a c++ project tcp_echo project wth lwip& freeRTOS middleware and added the mongoose embedded network stack following this [ref](https://mongoose.ws/documentation/tutorials/nxp/nxp-mimxrt1020-freertos/)
2. Then, added the gitclone repo of ArduinoJson into directory and added its path in include. There were duplication due to rerad/write/accept/bind calls in ArduinoJson & lwip's socket.h file. Hence, I've renamed those in lwip's socket.h file alone - and made corresponding name changes in lwip depended other files too..
3. Then, added the gitclone repo of MicroOCPP into directory and added its path in include. Then passed these flags - MO_PLATFORM_NONE -MO_CUSTOM_CONSOLE -MO_CUSTOM_TIMER -MO_CUSTOM_WS -MO_PLATFORM=MO_PLATFORM_NONE -MOCPP_DEACTIVATE_FLASH
4. Then, added the gitclone repo of MicroOcppMongooseClient into directory and added its path in include. 

Now, in main source file, I've added the following, there is no error - still some warnnings are there !!
#include "mongoose.h"
#include "ArduinoJson.h"
#include "MicroOcpp.h"
#include "MicroOcppMongooseClient.h"

I'm certain the process is not completed. As, my lwip example uses ETH interface, while the Connection.h uses WiFi as of now. I've to edit those. Yet, till now, am I on right track? If not kindly let me know @matth-x