matth-x/MicroOcpp #199: Send MeterValues

NuriaParra8997, 2023-09-29:
Hello,

I'm trying to send meter values using **ocpp_addMeterValueInput** in C, but I don't know how to use this function. 
Does anyone know how to use this function?

matth-x, 2023-09-29:
The MeterValueInput is a callback function which returns the current meter reading when it is executed. For example:

```cpp
//this register is always 42Wh, but in a real world scenario, the electricity meter would update it
float dummy_energy_register = 42.f;

//defined in your user code, but the library will call it later to fetch the energy register value
float get_energy_callback() {
    return dummy_energy_register;
}

void app_main() {
    //setup code ...

    ocpp_addMeterValueInput(get_energy_callback, "Energy.Active.Import.Register", "Wh", NULL, NULL);

    //...
}
```

When the library is configured with a MeterValueInput, it will automatically send MeterValues messages regularly. To test this, start a transaction and wait for 60 seconds.

Feel free to reopen if you have follow-up questions.

closed 2023-09-29 by matth-x

ap9389552, 2024-10-18:
hi @matth-x 
[image]