matth-x/MicroOcpp #200: setOnResetExecute - callback after 15 seconds + Persistent state

shardul10, 2023-09-29:
The callback I have registered with setOnResetExecute() is getting called 15 seconds after receiving Reset.  The library is sending StatusNotification - Unavailable immediately. 

29.09.2023 13:22:25 INF [sendTXT] [2,"1000004","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Unavailable","timestamp":"2023-09-29T13:22:25.888Z"}]
29.09.2023 13:22:25 INF [operator()] [3,"1000004",{}]
29.09.2023 13:22:25 INF [sendTXT] [2,"1000005","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Unavailable","timestamp":"2023-09-29T13:22:25.889Z"}]
29.09.2023 13:22:26 INF [operator()] [3,"1000005",{}]
**29.09.2023 13:22:40 ERR [EVSE_reset] Resetting EVSE**

Looks like it's by design, but any specific reason for 15 seconds delay in calling the callback?

Also as per OCPP 1.6 the state of connector state should persist. So a reset after connector is put in Unavailable state should remain keep the state Unavailable, right? Even after reboot?

matth-x, 2023-09-29:
There's no specific reason that it's exactly 15 seconds, but some delay is necessary to ensure that the Reset.conf and the subsequent StatusNotification can be sent out.

When the server makes the connector Unavailable via ChangeAvailability command, then this setting should persist over reboots. After a Reset command, the connector becomes Unavailable too, but it keeps this state only until the charger performs the reset successfully or aborts it. All that behavior is handled by the library.

Feel free to reopen if you have follow-up question.

closed 2023-09-29 by matth-x

shardul10, 2023-10-02:
I also registered setOnResetNotify() callback to get immediate notification.