matth-x/MicroOcpp #201: Persistent State after Reset

shardul10, 2023-09-30:
I couldn't reopen the issue [#200](url) hence creating a new one. 

This is what I tried.

1. BootNotification sent and Accepted
2. CMS sends ChangeAvailability (Inoperative) 
3. Reset

After Reset completes (ESP.restart()), new BootNotification is sent and Accepted, but the state is now Available.

30.09.2023 04:48:42 INF [operator()] [2,"e42c2d33-fc12-40a6-9ad8-aa5ebcabbbc1","ChangeAvailability",{"connectorId":0,"type":"Inoperative"}]
30.09.2023 04:48:43 INF [sendTXT] [3,"e42c2d33-fc12-40a6-9ad8-aa5ebcabbbc1",{"status":"Accepted"}]
30.09.2023 04:48:43 INF [sendTXT] [2,"1000003","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Unavailable","timestamp":"2023-09-30T04:48:43.016Z"}]
30.09.2023 04:48:43 INF [operator()] [3,"1000003",{}]
30.09.2023 04:48:43 INF [sendTXT] [2,"1000004","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Unavailable","timestamp":"2023-09-30T04:48:43.017Z"}]
30.09.2023 04:48:43 INF [operator()] [3,"1000004",{}]
30.09.2023 04:48:56 INF [operator()] [2,"a7892059-3c31-49ad-ab7f-a0c3034ad925","Reset",{"type":"Hard"}]
30.09.2023 04:48:56 INF [sendTXT] [3,"a7892059-3c31-49ad-ab7f-a0c3034ad925",{"status":"Accepted"}]
30.09.2023 04:49:11 ERR [EVSE_reset] Resetting EVSE
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
30.09.2023 04:49:14 INF [sendTXT] [2,"1000000","BootNotification",{"chargeBoxSerialNumber":"1234","chargePointModel":"TBD_EVSE_1.0","chargePointSerialNumber":"1234","chargePointVendor":"TBD","firmwareVersion":"0.0.5","iccid":"","imsi":"","meterSerialNumber":"","meterType":""}]
30.09.2023 04:49:14 INF [operator()] [3,"1000000",{"status":"Accepted","currentTime":"2023-09-30T04:49:13.799Z","interval":14400}]
30.09.2023 04:49:15 INF [sendTXT] [2,"1000001","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2023-09-30T04:49:14.296Z"}]
30.09.2023 04:49:15 INF [operator()] [3,"1000001",{}]
30.09.2023 04:49:15 INF [sendTXT] [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2023-09-30T04:49:14.297Z"}]
30.09.2023 04:49:15 INF [operator()] [3,"1000002",{}]

Is this an expected behaviour or should the Charge Point remain in UnAvailable state?

matth-x, 2023-09-30:
I checked this behavior with the Simulator and it's intact there. Probably the filesystem access doesn't work in your case. Maybe the log will reveal more hints when the verbosity level is increased to DEBUG. You can increase the verbosity level in platformio.ini, by changing the following line into `-D MOCPP_DBG_LEVEL=MOCPP_DL_DEBUG`:

https://github.com/matth-x/MicroOcpp/blob/38f1e6bd629ccbae35ede7e134b794d7e8a170c3/platformio.ini#L33

shardul10, 2023-09-30:
Here’s the log file for this scenario. Hope it helps. 

￼

matth-x, 2023-10-01:
The log message

```
[OCPP] debug (ConfigurationContainerFlash.cpp:30): Populate FS: create configuration file
```

says that no configuration file has been found and that the library populates a new file instead. That's normal for the first run on a fresh controller but afterwards this message should disappear. There was a bug in the Arduino LittleFS adapter when I migrated to the LittleFS lib included in arduino-esp32. dea20fd81b623eb77e282331609d94c83e5a47b2 fixes this. Upgrading to the latest commit in the develop branch could resolve the issue.

shardul10, 2023-10-02:
Upgrading to the latest commit on develop branch did resolve this.

closed 2023-10-02 by shardul10