matth-x/MicroOcpp #202: Reset during charging

shardul10, 2023-10-02:
I am trying a scenario where Steve OCPP sends a Reset during Charging. I see the reset is triggering a StopTransaction with idTag "HardReset" which is not liked by Steve (invalid idTag). Also there is an additional StatusNotification - Unavailable with error code "LocalListConflict" which I don't understand. Can you please take a look?

```
02.10.2023 05:50:36 INF [operator()] [2,"5d8d790d-6f60-4cc4-9a2d-51717e76bd7f","RemoteStartTransaction",{"idTag":"9010239100"}]
02.10.2023 05:50:37 INF [sendTXT] [3,"5d8d790d-6f60-4cc4-9a2d-51717e76bd7f",{"status":"Accepted"}]
02.10.2023 05:50:37 INF [sendTXT] [2,"1000003","StartTransaction",{"connectorId":1,"idTag":"9010239100","meterStart":0,"timestamp":"2023-10-02T05:50:37.586Z"}]
02.10.2023 05:50:37 INF [operator()] [3,"1000003",{"transactionId":19,"idTagInfo":{"status":"Accepted","expiryDate":"2023-10-02T06:50:37.889Z"}}]
02.10.2023 05:50:38 INF [sendTXT] [2,"1000004","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2023-10-02T05:50:37.831Z"}]
02.10.2023 05:50:38 INF [operator()] [3,"1000004",{}]
02.10.2023 05:50:48 INF [operator()] [2,"5d7bf710-7485-478e-a350-6e96d09938dd","Reset",{"type":"Hard"}]
02.10.2023 05:50:48 ERR [EVSE_stop] Stopping EVSE
02.10.2023 05:50:48 INF [sendTXT] [3,"5d7bf710-7485-478e-a350-6e96d09938dd",{"status":"Accepted"}]
02.10.2023 05:50:49 INF [sendTXT] [2,"1000005","StopTransaction",{"idTag":"**HardReset**","meterStop":22,"timestamp":"2023-10-02T05:50:48.813Z","transactionId":19}]
02.10.2023 05:50:49 INF [operator()] [3,"1000005",{"idTagInfo":{"status":"Invalid"}}]
02.10.2023 05:50:49 INF [sendTXT] [2,"1000006","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Unavailable","timestamp":"2023-10-02T05:50:49.177Z"}]
02.10.2023 05:50:49 INF [operator()] [3,"1000006",{}]
02.10.2023 05:50:49 INF [sendTXT] [2,"1000007","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Unavailable","timestamp":"2023-10-02T05:50:49.179Z"}]
02.10.2023 05:50:49 INF [operator()] [3,"1000007",{}]
02.10.2023 05:50:49 INF [sendTXT] [2,"1000008","StatusNotification",{"connectorId":0,"errorCode":"**LocalListConflict**","status":"Unavailable","timestamp":"2023-10-02T05:50:49.342Z"}]
02.10.2023 05:50:49 INF [operator()] [3,"1000008",{}]
02.10.2023 05:51:03 ERR [EVSE_reset] Restarting EVSE
```

shardul10, 2023-10-02:
The second part of StatusNofification with LocalListConflict error is probably due to some mismatch in older list I was trying to send from Steve OCPP. The idTag of HardReset is still there.

shardul10, 2023-10-02:
In order to clear some stale local list and to clear some states, I had to do a complete flash format. I guess some stale information remained in the files and was not getting cleared. Is there is mocpp_reset() type function that will delete all files/states and initialise mocpp to clean slate? May not be required in field, but I think extremely useful during testing.

shardul10, 2023-10-02:
> The second part of StatusNofification with LocalListConflict error is probably due to some mismatch in older list I was trying to send from Steve OCPP. The idTag of HardReset is still there.

I think I found the reason why idTag is HardReset. In Reset.cpp the code below is called during processing or reset. The first parameter should have been nullptr, right?

`model.getConnector(cId)->endTransaction(isHard ? "HardReset" : "SoftReset");
`
I made this change and tested it successfully. 

`model.getConnector(cId)->endTransaction(nullptr, isHard ? "HardReset" : "SoftReset");
`

matth-x, 2023-10-02:
@shardul10 Thanks for the report! I'm confirming this, it's a bug and I will resolve it very soon.

In this library, the "ClearCache" operation also clears the transaction journal and after a Reset, the transaction state should be fresh again. If you need a programmatic way, you can take the following lines as an example:

https://github.com/matth-x/MicroOcpp/blob/b0f416c5c92a10df85c96a446f8dcf13fc550d06/src/MicroOcpp.cpp#L150-L156

This example even goes a little further and instead of cleaning only the transaction journal (all files starting with "sd", "tx" and "op"), it also clears the reservation list and all charging schedules (files starting with "sc-").

shardul10, 2023-10-05:
Thanks for the fix. Assume you will close this.

commit cce3d815 (2023-10-02): fix endTransaction params (fixes #202)
closed 2023-10-22 by matth-x