matth-x/MicroOcpp #203: Handling fault during idle and charging

shardul10, 2023-10-02:
I tested a scenario of charger faulting while idle. The OCPP library sends StatusNotification - Unavailable (Faulted) to CMS.  
The fault was reported through addErrorCodeInput() by returning "InternalError".

When I faulted the charger during charging, the same StatusNotification - Unavailable (Faulted) was sent, but the charging continued. 

```
02.10.2023 06:56:00 INF [operator()] [2,"8d2a0abc-8a9b-437f-82b6-6da8cba9441c","RemoteStartTransaction",{"connectorId":1,"idTag":"7507344113"}]
02.10.2023 06:56:00 INF [sendTXT] [3,"8d2a0abc-8a9b-437f-82b6-6da8cba9441c",{"status":"Accepted"}]
02.10.2023 06:56:00 INF [sendTXT] [2,"1000003","StartTransaction",{"connectorId":1,"idTag":"7507344113","meterStart":0,"timestamp":"2023-10-02T06:56:00.363Z"}]
02.10.2023 06:56:00 INF [operator()] [3,"1000003",{"transactionId":25,"idTagInfo":{"status":"Accepted","expiryDate":"2023-10-02T07:56:00.860Z"}}]
02.10.2023 06:56:01 INF [sendTXT] [2,"1000004","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Charging","timestamp":"2023-10-02T06:56:00.789Z"}]
02.10.2023 06:56:01 INF [operator()] [3,"1000004",{}]
02.10.2023 06:56:21 ERR [EVSE_loop] Heartbeat not received for 30 seconds...faulting
02.10.2023 06:56:21 INF [sendTXT] [2,"1000005","StatusNotification",{"connectorId":1,"errorCode":"InternalError","status":"**Faulted**","timestamp":"2023-10-02T06:56:21.410Z"}]
02.10.2023 06:56:21 INF [operator()] [3,"1000005",{}]
02.10.2023 06:56:30 INF [sendTXT] [2,"1000006","MeterValues",{"connectorId":1,"transactionId":25,"meterValue":[{"timestamp":"2023-10-02T06:56:30.742Z","sampledValue":[{"value":"62.1","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"7360.0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"}]}]}]
02.10.2023 06:56:30 INF [operator()] [3,"1000006",{}]
02.10.2023 06:57:00 INF [sendTXT] [2,"1000007","MeterValues",{"connectorId":1,"transactionId":25,"meterValue":[{"timestamp":"2023-10-02T06:57:00.742Z","sampledValue":[{"value":"123.5","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"7360.0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"}]}]}]
02.10.2023 06:57:00 INF [operator()] [3,"1000007",{}]
02.10.2023 06:57:30 INF [sendTXT] [2,"1000008","MeterValues",{"connectorId":1,"transactionId":25,"meterValue":[{"timestamp":"2023-10-02T06:57:30.742Z","sampledValue":[{"value":"184.8","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"7360.0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"}]}]}]
02.10.2023 06:57:30 INF [operator()] [3,"1000008",{}]
02.10.2023 06:58:00 INF [sendTXT] [2,"1000009","MeterValues",{"connectorId":1,"transactionId":25,"meterValue":[{"timestamp":"2023-10-02T06:58:00.742Z","sampledValue":[{"value":"246.2","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"7360.0","context":"Sample.Periodic","measurand":"Power.Active.Import","unit":"W"}]}]}]
02.10.2023 06:58:00 INF [operator()] [3,"1000009",{}]
```

How to handle this scenario? Should I initiate a StopTransaction() after detecting the fault (unlike Reset where the StopTransaction is sent by library)?

shardul10, 2023-10-02:
I added a check for existing transaction/state while reporting fault in addErrorCodeInput() and if active, stopping the transaction with `endTransaction(nullptr, "EmergencyStop"); `

Couldn't find any better reason in the specification. This seems to be working, but is this the correct way of handling fault during transaction?

matth-x, 2023-10-02:
Yes, it is the correct way. Some error scenarios are only temporary and the transaction should continue when the error disappeared (e.g. "HighTemperature"). Other errors are more severe and the charging should stop immediately. It's up to the vendor to decide how to handle error scenarios, OCPP doesn't make any rules about it.

Putting `endTransaction(nullptr, ocppErrorReason)` into the ErrorCode Input is the best practice here.

shardul10, 2023-10-02:
Thanks. Closing this now.

closed 2023-10-02 by shardul10