matth-x/MicroOcpp #205: Checking beginTransaction() status - success or failure

shardul10, 2023-10-03:
I am following one of the examples to use beginTransaction() to Authorise and issue StarTransaction. The API returns immediately if it is able to initiate the transaction.

How to check:
1. if Authorise was successful or not
2. if StartTransaction was sent or not

I didn't see any callbacks to register to get this information. These triggers are required to inform the charger to start/stop charing.

shardul10, 2023-10-03:
I guess I can use setTxNotificationOutput() to get the status? So if I get MicroOcpp::TxNotification::Authorized, I can instruct the charger to offer energy and as soon as connector is plugged (setConnectorPluggedInput()), library will send StartTransaction. Is that correct understanding?

matth-x, 2023-10-03:
The following functions return the current transaction state:

https://github.com/matth-x/MicroOcpp/blob/38f1e6bd629ccbae35ede7e134b794d7e8a170c3/src/MicroOcpp.h#L154-L175

In addition to that, `beginTransaction(...)` and `getTransaction(...);` return a pointer to the current transaction process (if existent). The [Transaction](https://github.com/matth-x/MicroOcpp/blob/38f1e6bd629ccbae35ede7e134b794d7e8a170c3/src/MicroOcpp/Model/Transactions/Transaction.h) class contains all data related to the transaction.

`ocppPermitsCharge(...)` returns if charging is permitted by OCPP at the moment. This is not always the same as `isTransactionRunning()`, because for example, a transaction can be DeAuthorized and continue running while charging is not allowed anymore. Here is the API description:

https://github.com/matth-x/MicroOcpp/blob/38f1e6bd629ccbae35ede7e134b794d7e8a170c3/src/MicroOcpp.h#L199-L205

The functions above return the current status at any time. The library also emits transaction-related events defined in the [TxNotification](https://github.com/matth-x/MicroOcpp/blob/38f1e6bd629ccbae35ede7e134b794d7e8a170c3/src/MicroOcpp/Model/ConnectorBase/Notification.h) enum (it's called "notification" to avoid confusion with the transaction events which will be introduced for OCPP 2.0.1). The OpenEVSE integration shows how the TxNotification Output can be used to show messages on the display:

https://github.com/OpenEVSE/openevse_esp32_firmware/blob/8c9a482c912005f4c5f0e2d82d2f596e665dd6eb/src/ocpp.cpp#L328-366

But these TxNotifications shouldn't be the basis of controlling the high power electronics (use `ocppPermitsCharge(...)` instead). In case of a blackout the library won't emit the latest TxNotification again and the API doesn't guarantee that a TxNotification is always sent in case of a Start-/StopTransaction. TxNotifications are more relevant when the reason of an authorization failure is of interest.

shardul10, 2023-10-04:
Thanks for detailed response. The problem is where to check the status of transaction and for how long (in case of timeout)? Here's the scenario:

> In the loop() I get RFID and issue beginTransaction().
> Now I have to know the status of authorisation first

Do I start checking the status immediately (using the APIs you have suggested) until I get some status update? How long? That would block the loop(). 

I looked at the Open-evse example where the beginTransaciton() is issued similar to my code in onIdTagInput(). Where does it check the success/failure to start offering energy?

Would you consider adding a callback(s) to beginTransaction() for authorisation and starttransaction responses? Alternatively adding a setOnReceiveConf() similar to setOnSendConf()? 

I think it will greatly simplify usage of beginTransaction().

matth-x, 2023-10-04:
Would this work for you?

```cpp
//tracking variable for detecting status changes
bool trackOcppPermitsCharge = false;

void loop() {
    //...
    if (trackOcppPermitsCharge != ocppPermitsCharge()) {
        //status change, enable / disable the PWM communication (and relay locks) here
    }
    trackOcppPermitsCharge = ocppPermitsCharge();
}
```

The OpenEVSE integration ultimately follows the same approach (here it polls `ocppPermitsCharge()`): https://github.com/OpenEVSE/openevse_esp32_firmware/blob/8c9a482c912005f4c5f0e2d82d2f596e665dd6eb/src/ocpp.cpp#L420-L424

shardul10, 2023-10-04:
I  will certainly try this approach. If the authorisation/reservation check fails inside beginTransaction(), ocppPermitsCharge() will never return true, right?

It would be good to inform the user that authentication failed. Should I continue to do that through TxNotification?

Will this function also return true in RemoteStartTransaction() scenario (once StartTransaction is sent)?

Thanks
Shardul

matth-x, 2023-10-04:
Yes, yes and yes :)

shardul10, 2023-10-04:
One more clarification please. 

Will the ocppPermitsCharge() return true only after setConnectorPluggedInput() returns true?

shardul10, 2023-10-05:
Closing this now. Using ocppPermitsCharge() to start energy offer.

closed 2023-10-05 by shardul10