matth-x/MicroOcpp #208: ESP gets rebooted when getDiagnostics was called from CMS

VjSargunam, 2023-10-09:
Hi Matt,

I tried to get the getdiagnostics.log file from the OCPP client. I initiated the GetDiagnostics request from CMS as below.
`[2,"815b4363-d702-4931-971d-e557bf2e38ea","GetDiagnostics",{"location":"ftp://<username>:<password>@rrthost.myftp.org","retries":1,"retryInterval":1,"startTime":"2023-10-09T09:25:23.000Z","stopTime":"2023-10-09T09:25:23.000Z"}]. `
                But when I initiate this request the esp32 gets rebooted. is this the correct way to initiate the request. please clarify.And this is my CMS requesting format for location `ex:ftp://ftpusername:password@ftpserverIP/locationUrl/DiagnosticsFileName.gpg`

Regards
VJSargunam

matth-x, 2023-10-10:
Hi @VjSargunam,

At the moment, there's no integrated logging / file upload mechanism. The OCPP specs also lack a description of what should be part of the diagnostics file and I believe that the vendor should decide it, so I left the diagnostics functionality very minimal but customizable.

To implement GetDiagnostics, the OCPP lib needs to be configured with a custom upload handler and an upload status input. When the server sends a GetDiagnostics request to the charger, then the OCPP lib executes the upload handler and forwards all parameters to it. The custom upload handler needs to implement the generation of the diagnostics data and the actual file upload. During the process, the OCPP lib polls the upload status input so that it can inform the server about the current progress via DiagnosticsStatusNotification messages.

As an example, the OpenEVSE platform uses the HTTP protocol to upload its event history to the server: https://github.com/OpenEVSE/openevse_esp32_firmware/blob/8c9a482c912005f4c5f0e2d82d2f596e665dd6eb/src/ocpp.cpp#L480-L595

Currently I'm evaluating an FTP client based on Mongoose which I published in the [MicroOcppMongoose](https://github.com/matth-x/MicroOcppMongoose) repository with very promising results. I'm planning to release an application reference very soon.

Btw. and thank you very much for the bug report. Of course the library shouldn't crash when there's no diagnostics handler. This is already fixed in the develop branch.

VjSargunam, 2023-10-10:
Hi Matt,

              How to add this function void ArduinoOcppTask::initializeDiagnosticsService() {} in the main program since it was not present in latest version of MicroOcpp

matth-x, 2023-10-14:
To configure the library with the custom upload handler in main.cpp, fetch the DiagnosticsService object and pass the upload handler to it like this:

```cpp
// your upload handler
bool customUploadHandler(const std::string &location, ArduinoOcpp::Timestamp &startTime, ArduinoOcpp::Timestamp &stopTime) {
    //TODO collect your hardware data here

    //TODO upload the diagnostics file to `location`

    return true;
}

void setup() {
    // initialize the charger and MicroOcpp here

    // fetch the DiagnosticsService object and configure it
    MicroOcpp::DiagnosticsService *diagService = getDiagnosticsService();
    if (diagService) {
        // set the custom upload handler defned above which the library will call now as soon as a GetDiagnostic request comes in
        diagService->setOnUpload(customUploadHandler);
    }

}
```

Generally, the OpenEVSE reference integration of GetDiagnostics is still up to date with the master branch (apart from the project name change from ArduinoOcpp to MicroOcpp). Especially, it shows how collecting the diagnostics data and uploading it to the server can look like.

VjSargunam, 2023-10-16:
Hi Matt,
            Thanks for your reply. Is this scenario being same for firmware update too? Or is there is any other approach to do the same?

matth-x, 2023-10-16:
Yes, the approach is the same.

Again, the OpenEVSE integration is a reference for the custom FW update. On top of that, the OCPP lib also includes a very simple demo FW update handler (https://github.com/matth-x/MicroOcpp/blob/master/src/MicroOcpp/Model/FirmwareManagement/FirmwareService.cpp#L336-L369). It works for the ESP platforms if using the default links2004 WebSocket lib. Furthermore, the UpdateFirmware request needs to point to an http:// location.

VjSargunam, 2023-10-31:
Hi Matt,
           As per your suggestion I included this piece of code `MicroOcpp::DiagnosticsService *diagService = DiagnosticsService();` in the setup() function to initialize getdiagnostic service but iam getting error like identifier `"DiagnosticsService" is undefined` while compile the code. I even added the header file `#include <MicroOcpp/Model/Diagnostics/DiagnosticsService.h>` in the main.cpp but still iam getting the above error. plz clarfy

matth-x, 2023-10-31:
Oh this function is excluded by an `#if` preprocessor statement and needs to be enabled first (probably I will change it to enabled by default in a future version). Set the build flag `MO_CUSTOM_DIAGNOSTICS` (or `MOCPP_CUSTOM_DIAGNOSTICS`, depending on your version), in the platformio.ini and then it should be there.