matth-x/MicroOcpp #211: EndTransaction with transactionId

drxroot, 2023-10-12:
How can I close the transaction with the transaction ID given to us by CSMS when I initiate a transaction? Because sometimes the internet connection drops and I cannot finish the charging process. Since I cannot end the charging process, I cannot start a new charging process.

matth-x, 2023-10-14:
The function `endTransaction()` terminates the currently active transaction immediately. It is declared in MicroOcpp.h:

https://github.com/matth-x/MicroOcpp/blob/d486288d5a151d271bf4e78a534eda8231e99381/src/MicroOcpp.h#L152

After the transaction process has been ended, the library is ready to begin a new transaction.

This library caches the transaction data locally to handle offline scenarios. In case you end a transaction while the charger is offline, it will retry sending the StopTransaction once the charger gets online again. No need to check for the current internet connection, in any case you can call `endTransaction()` as soon as charging is finished.

drxroot, 2023-10-27:
Thanks for the answer @matth-x . So, when ESP is reset, how can I continue sending x value again or turn it off? Because when it is reset, I cannot close transaction with endTransaction().

drxroot, 2023-10-27:
Thanks for the answer. So, when ESP is reset, how can I continue sending Wh value again or turn it off? Because when it is reset, I cannot close it with endTransaction() and update Wh value setEnergyMeterInput() or addMeterValueInput().

matth-x, 2023-10-27:
In case the OCPP lib forgets about the current transaction after the ESP has reset, then most likely the filesystem access doesn't work. Are there suspicious log messages about this?

kuribox, 2023-10-30:
I have a similar issue, before ESP reboots or power losses. Is there any way to terminate the active transaction when the ESP restarts, either without a transactionId, or with the last cached transaction id?

matth-x, 2023-10-31:
Before the ESP restarts, at the time when the power loss occurs, it's complicated. The typical approach is more to terminate any active transaction when the ESP boots up again and that will also deal with transactions interrupted by power losses. When initializing the OCPP lib, you can place `endTransaction(nullptr, "PowerLoss");` somewhere after `mocpp_initialize` and after setting the main energy meter. It's always safe to call `endTransaction`, even if there's no active transaction.

maxikys, 2024-01-20:
Hello every one! I  follow **matth-x** and it is solve problem. But I don't to know to provide correct value of energy in parameter _meterStop_.  **matth-x**, please explain how I can to do this.

matth-x, 2024-01-21:
The library will fetch the energy register from the EnergyMeterInput once it creates the StopTransaction request. In the initialization phase of the charger you need to set the callback function EnergyMeterInput. After that, the library will execute the EnergyMeterInput and take its return value as the meterStop parameter.

See this example of setting the EnergyMeterInput:

https://github.com/matth-x/MicroOcpp/blob/12d4868200d9675d98ff81b9c226e5deace64c12/examples/ESP/main.cpp#L66-L69

MicroOcpp will execute this lambda when sending the StopTransaction. Now the lambda only returns 0, but you can change it so that it will return the actual energy register.