matth-x/MicroOcpp #212: [OCPP] info (Connection.cpp:54): Disconnected

d6e6t6a6l6e6s, 2023-10-13:
I flash MicroOcpp (main.cpp) to ESP-WROOM-32. ESP connects to Wi-Fi successfully but won't connect to my SteVe server at all. The "Etrel Inch Pro" charging station connects to the SteVe without any problems. What are you doing wrong? Here are my the configurations:

**steve "main.properties":**
```
# Jetty configuration
#
server.host = 192.168.77.28
server.gzip.enabled = true
# Jetty HTTP configuration
#
http.enabled = true
http.port = 6226
```
**main.cpp:**
```
#define OCPP_HOST "occp-server"
#define OCPP_PORT 6226
#define OCPP_URL "ws://192.168.77.28/steve/websocket/CentralSystemService/espdemo"
```
**ESP Serial Output:**
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[main] Wait for WiFi: . connected!
[OCPP] info (MicroOcpp.cpp:387): Added ConnectorPluggedInput. Transaction-management is in auto mode now
[OCPP] info (Connection.cpp:54): Disconnected
[OCPP] info (Connection.cpp:54): Disconnected
[OCPP] info (Connection.cpp:54): Disconnected
```

priya2212, 2023-10-25:
You need to add port also in the OCPP_URL.

d6e6t6a6l6e6s, 2023-11-17:
You mean that?

**main.cpp:**
```
#define OCPP_HOST "occp-server"
#define OCPP_PORT 6226
#define OCPP_URL "ws://192.168.77.28:6226/steve/websocket/CentralSystemService/espdemo"
```
I am getting the same error..

rafaelcf03, 2024-03-15:
Hey! How you doing?
I'm facing the same problem. Have you solved? What should I do?
Thanks already!