matth-x/MicroOcpp #214: Ethernet Support with OCPP

priya2212, 2023-10-17:
Dear @matth-x , 
Thank you for making the resources publicly, it is definitely helping us alot for development. 
I have one small query - 
Can we connect the OCPP with ethernet support? I am able to connect the OCPP with esp32 and wifi. 
Platform Used - VSCOde, PlatformIO and framework - Arduino, Development board - ESP32-Wroom-32

When I am connecting the device with ethernet and requesting it to connect it to OCPP server, facing below issues - 
Case 1 - with server **ws://echo.websocket.events/** - Error - hostByName(): DNS Failed for echo.websocket.events
Case 2 - with steve server - Error - the program keeps crashing with error - assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Would be great if you can give some valuable insights. The development is on hold and ethernet is the primary requirement.

priya2212, 2023-10-17:
Dear @matth-x , Many thanks for your quick and positive response. 
I have tried both the solutions but nothing worked out, facing same crashing error. -  Tried adding delay at multiple locations and commenting the sendTXT line. 

I have tried to use Ethernet code separately without Ocpp, it works perfectly fine. 
PS - I am not using Mongoose WS Adapter, just using Ethernet Adapter library. 

Below is the setup code - 
```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  Serial.print(F("[main] Wait for Ethernet: "));

//start the Ethernet connection:
    delay(1000);
    Ethernet.init(5);   // MKR ETH Shield
    delay(5000);

    if (Ethernet.begin(mac)) { // Dynamic IP setup
        Serial.println("DHCP OK!");
    }else{
        Serial.println("Failed to configure Ethernet using DHCP");
        // Check for Ethernet hardware present
        if (Ethernet.hardwareStatus() == EthernetNoHardware) {
          Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
          while (true) {
            delay(1); // do nothing, no point running without Ethernet hardware
          }
        }
        if (Ethernet.linkStatus() == LinkOFF) {
          Serial.println("Ethernet cable is not connected.");
        }
    }
    delay(5000);
 
 
    Serial.print("Local IP : ");
    Serial.println(Ethernet.localIP());
    Serial.print("Subnet Mask : ");
    Serial.println(Ethernet.subnetMask());
    Serial.print("Gateway IP : ");
    Serial.println(Ethernet.gatewayIP());
    Serial.print("DNS Server : ");
    Serial.println(Ethernet.dnsServerIP());
 
   Serial.println("Ethernet Successfully Initialized");
  Serial.println(F(" connected!"));

  delay(5000);

    /*
     * Initialize the OCPP library
     */
    mocpp_initialize(OCPP_HOST, OCPP_PORT, OCPP_URL, "OCPP Station", "STABOR EVSE");

    setSmartChargingPowerOutput([](float limit) {
        //set the SAE J1772 Control Pilot value here
        Serial.printf("[main] Smart Charging allows maximum charge rate: %.0f\n", limit);
    });

    setConnectorPluggedInput([]() {
        //return true if an EV is plugged to this EVSE
        return true;
    });
}
```

```
void loop() {

    /*
     * Do all OCPP stuff (process WebSocket input, send recorded meter values to Central System, etc.)
     */
    mocpp_loop();
}
```

I have enabled the exception analyser to backtrace the exception, below are the logs - 
```
[main] Wait for Ethernet: DHCP OK!
Local IP : 192.168.137.162
Subnet Mask : 255.255.255.0
Gateway IP : 192.168.137.1
DNS Server : 192.168.137.1
Ethernet Successfully Initialized
 connected!
test
[OCPP] info (MicroOcpp.cpp:393): Added ConnectorPluggedInput. Transaction-management is in auto mode now

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace: 0x400835ad:0x3ffc6fb0 0x4008a279:0x3ffc6fd0 0x4008f559:0x3ffc6ff0 0x4010ef71:0x3ffc7120 0x4011d875:0x3ffc7150 0x4011d8d5:0x3ffc7170 0x4010ec9d:0x3ffc71c0 0x400fa741:0x3ffc71e0 0x400f9ece:0x3ffc7250 0x400fdf22:0x3ffc7280 0x400d9025:0x3ffc72a0 0x400d93bf:0x3ffc72c0 0x400d2e04:0x3ffc72e0 0x400f9a83:0x3ffc7300 0x401026b9:0x3ffc7320

  #0  0x400835ad:0x3ffc6fb0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008a279:0x3ffc6fd0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4008f559:0x3ffc6ff0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x4010ef71:0x3ffc7120 in tcpip_send_msg_wait_sem at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:455 (discriminator 1)
  #4  0x4011d875:0x3ffc7150 in netconn_apimsg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:136
  #5  0x4011d8d5:0x3ffc7170 in netconn_new_with_proto_and_callback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:166
  #6  0x4010ec9d:0x3ffc71c0 in lwip_socket at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/sockets.c:1774
  #7  0x400fa741:0x3ffc71e0 in WiFiClient::connect(IPAddress, unsigned short, int) at /Users/admin/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/sockets.h:656
      (inlined by) WiFiClient::connect(IPAddress, unsigned short, int) at /Users/admin/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:216
  #8  0x400f9ece:0x3ffc7250 in WiFiClient::connect(char const*, unsigned short, int) at /Users/admin/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:301
  #9  0x400fdf22:0x3ffc7280 in WebSocketsClient::loop() at .pio/libdeps/esp32doit-devkit-v1/WebSockets/src/WebSocketsClient.cpp:266
  #10 0x400d9025:0x3ffc72a0 in MicroOcpp::EspWiFi::WSClient::loop() at src/MicroOcpp/Core/Connection.cpp:41
  #11 0x400d93bf:0x3ffc72c0 in MicroOcpp::Context::loop() at src/MicroOcpp/Core/Context.cpp:26
  #12 0x400d2e04:0x3ffc72e0 in mocpp_loop() at src/MicroOcpp.cpp:255
  #13 0x400f9a83:0x3ffc7300 in loop() at src/main.cpp:141
  #14 0x401026b9:0x3ffc7320 in loopTask(void*) at /Users/admin/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```
I am also using W5500 module with the ESP32-wroom module. As you have also encountered the same problem and solved it, I am hopeful that I will be able to find some solution with your knowledge, please help.

matth-x, 2023-10-18:
Cool, that clean code base is definitely helpful for tracking the issue down.

The following workaround helped me with a similar issue. It is based on the other, more explicit initialization approach shown in the [ESP-IDF example](https://github.com/matth-x/MicroOcpp/blob/master/examples/ESP-TLS/main.cpp). The explicit initialization of the WebSocket library gives more control over the WebSocket connection. You can remove all TLS for the moment, but the W5500 integration generally looks good and should work the same in the ESP-IDF example.

Now, that you have the `wsockSecure` and the `osockSecure` objects, you can add the following delay into `loop()`:

```cpp
void loop() {

    if (osockSecure.getLastRecv() == 0) {
        //recv timestamp is still at 0 ms --> no WS has ever been received. Only loop WS lib and not OCPP lib
        wsockSecure.loop();
        return;
    }

    mocpp_loop();
}
```

Can you confirm if this works for you?

priya2212, 2023-10-18:
@matth-x Thank you again for giving your time and support for the problem statement. 
I have tried adding without TLS but unfortunately, the code keeps crashing in wsockSecure.loop();. 

Here are the logs - 
```
assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace: 0x400835ad:0x3ffc7150 0x4008a279:0x3ffc7170 0x4008f559:0x3ffc7190 0x4010b025:0x3ffc72c0 0x40119881:0x3ffc72f0 0x401198e1:0x3ffc7310 0x4010ad51:0x3ffc7360 0x400f6a39:0x3ffc7380 0x400f61c6:0x3ffc73f0 0x400fa196:0x3ffc7420 0x400f5edd:0x3ffc7440 0x400fe931:0x3ffc7460

  #0  0x400835ad:0x3ffc7150 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008a279:0x3ffc7170 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4008f559:0x3ffc7190 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x4010b025:0x3ffc72c0 in tcpip_send_msg_wait_sem at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:455 (discriminator 1)
  #4  0x40119881:0x3ffc72f0 in netconn_apimsg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:136
  #5  0x401198e1:0x3ffc7310 in netconn_new_with_proto_and_callback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:166
  #6  0x4010ad51:0x3ffc7360 in lwip_socket at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/sockets.c:1774
  #7  0x400f6a39:0x3ffc7380 in WiFiClient::connect(IPAddress, unsigned short, int) at /Users/admin/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/sockets.h:656
      (inlined by) WiFiClient::connect(IPAddress, unsigned short, int) at /Users/admin/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:216
  #8  0x400f61c6:0x3ffc73f0 in WiFiClient::connect(char const*, unsigned short, int) at /Users/admin/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:301
  #9  0x400fa196:0x3ffc7420 in WebSocketsClient::loop() at .pio/libdeps/esp32doit-devkit-v1/WebSockets/src/WebSocketsClient.cpp:266
  #10 0x400f5edd:0x3ffc7440 in loop() at src/main.cpp:158
  #11 0x400fe931:0x3ffc7460 in loopTask(void*) at /Users/admin/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```
Here is the code - 
```
#include <Arduino.h>

#include "MicroOcpp.h"
#include "MicroOcpp/Core/Configuration.h"
#include "Ethernet.h"
#include "SPI.h"

#include <MicroOcpp/Core/Connection.h> //need for setting TLS credentials

#define OCPP_HOST "192.168.102.65"
#define OCPP_PORT 8080
#define OCPP_URL "ws://192.168.102.65:8080/steve/websocket/CentralSystemService/ocpp_charger"

// #define OCPP_HOST "echo.websocket.events"
// #define OCPP_PORT 80
// #define OCPP_URL "ws://echo.websocket.events/"

bool flag = false;
float g_voltage = 0.f;

byte mac[] = {
  0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02
};

WebSocketsClient wsockSecure {};
MicroOcpp::EspWiFi::WSClient osockSecure {&wsockSecure};

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  Serial.print(F("[main] Wait for Ethernet: "));
  // start the Ethernet connection:
  delay(1000);
  Ethernet.init(5);   // MKR ETH Shield
  delay(1000);

  if (Ethernet.begin(mac)) { // Dynamic IP setup
    Serial.println("DHCP OK!");
  } else {
    Serial.println("Failed to configure Ethernet using DHCP");
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
      }
    }
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
  }
  delay(5000);

  Serial.print("Local IP : ");
  Serial.println(Ethernet.localIP());
  Serial.print("Subnet Mask : ");
  Serial.println(Ethernet.subnetMask());
  Serial.print("Gateway IP : ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("DNS Server : ");
  Serial.println(Ethernet.dnsServerIP());

  Serial.println("Ethernet Successfully Initialized");

  delay(5000);

  wsockSecure.begin(OCPP_HOST, OCPP_PORT, OCPP_URL, "ocpp_1.6");
  wsockSecure.setReconnectInterval(5000);
  wsockSecure.enableHeartbeat(15000, 3000, 2);
  mocpp_initialize(osockSecure, ChargerCredentials("My Charging Station", "My company name"));

  delay(5000);

}

void loop() {
  if (osockSecure.getLastRecv() == 0) {
    //recv timestamp is still at 0 ms --> no WS has ever been received. Only loop WS lib and not OCPP lib
    wsockSecure.loop();
    return;
  }
  mocpp_loop();
}

```

matth-x, 2023-10-18:
Does it change when you comment the line `mocpp_initialize(osockSecure, ChargerCredentials("My Charging Station", "My company name"));`?

priya2212, 2023-10-18:
@matth-x , No, the issue remains same. :(

matth-x, 2023-10-18:
Okay, without that line the OCPP lib is not executed at all, only the WebSocket lib. Probably it's easier to continue troubleshooting with the links2004/arduinoWebSocket examples. A further possibility would be to try the Mongoose WebSocket integration instead (actually it's even a little better tested than the default WS integration).

The principle of passing the Mongoose WS adapter to the OCPP lib is the same like you did in the ESP-TLS example. Also refer to MicroOcppSimulator to see how to initialize the Mongoose adapter. There are two things to watch out for:

- the Mongoose adapter needs to be allocated on runtime via a `new` statement
- the Mongoose sources (`mongoose.h` and `mongoose.c`) could be a little tricky to include. I'm not sure what the best practice is, but I ended up pasting them into `lib/mongoose/`

priya2212, 2023-10-19:
Dear @matth-x , 
you are right, the websocket library is not stable. I have checked with clean code based with arduinoWebSocket and the same problem arose. As per your recommendation, I switched to mongoose library with ESP-IDF framework and now, I see pretty stable connection with Ethernet and server. 
Thank you for your guidance. :) Closing the issue positively.

closed 2023-10-19 by priya2212

matth-x, 2023-10-22:
@priya2212, btw. thanks for sharing how you isolated and solved this issue! This helps me a lot to understand the limitations of the frameworks used here. Moreover, it will help a lot of people running into the same problem (and I will link your comment in future).

GanesanGuru, 2023-11-01:
Hi  @matth-x , @priya2212 
Is it possible to test this repo without the ESP32 modules rather in Laptop with Ubuntu OS? If so, kindly let me know how to obtain the same. 
PS : I've configure the Steve Server. I'm in need of client application, so that I can double understand the functionalities of ocpp.
Thanks in advance.

demirschmid61, 2024-06-11:
@matth-x @priya2212 

Thank you very much for the above communication, your efforts really helping me for the development

I am trying to OCPP client with ethernet and using **Arduino framework**. When I am trying to use **MicroOcppMongoose** with wifi it perfectly working as expected, but when trying to connect using ethernet, ESP32 is resetting.

**---------Logs when connected using ethernet----------**
Local IP : 192.168.0.238   
Subnet Mask : 255.255.255.0
Gateway IP : 192.168.0.2   
DNS Server : 192.168.0.14        
Ethernet Successfully Initialized

[MO] warning (MicroOcppMongooseClient.cpp:202): Insecure connection (WS)
assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace: 0x400836a5:0x3ffb1f10 0x4008a231:0x3ffb1f30 0x4008f679:0x3ffb1f50 0x401084d5:0x3ffb2080 0x401161ad:0x3ffb20b0 0x4011620d:0x3ffb20d0 0x401081d1:0x3ffb2120 0x400f504c:0x3ffb2140 0x400f4bb6:0x3ffb2160 0x400f8261:0x3ffb2180 0x400f837f:0x3ffb22e0 0x400f844e:0x3ffb2370 0x400f89ab:0x3ffb2410 0x400f91e6:0x3ffb2530 0x400d2ed9:0x3ffb25c0 0x400fba26:0x3ffb2820

  #0  0x400836a5:0x3ffb1f10 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x4008a231:0x3ffb1f30 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x4008f679:0x3ffb1f50 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x401084d5:0x3ffb2080 in tcpip_send_msg_wait_sem at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:455 (discriminator 1)
  #4  0x401161ad:0x3ffb20b0 in netconn_apimsg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:136
  #5  0x4011620d:0x3ffb20d0 in netconn_new_with_proto_and_callback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/api_lib.c:166
  #6  0x401081d1:0x3ffb2120 in lwip_socket at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/sockets.c:1774
  #7  0x400f504c:0x3ffb2140 in socket at C:/Users/atspc/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/sockets.h:656
      (inlined by) mg_socket_if_connect_tcp at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:3864
  #8  0x400f4bb6:0x3ffb2160 in mg_do_connect at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:3168
  #9  0x400f8261:0x3ffb2180 in mg_connect_opt at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:3345
  #10 0x400f837f:0x3ffb22e0 in mg_connect_http_base at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:8784
  #11 0x400f844e:0x3ffb2370 in mg_connect_ws_opt at .pio/libdeps/esp32-development-board/ArduinoMongoose/src/mongoose.c:10449
  #12 0x400f89ab:0x3ffb2410 in MicroOcpp::MOcppMongooseClient::maintainWsConn() at .pio/libdeps/esp32-development-board/MicroOcppMongoose/src/MicroOcppMongooseClient.cpp:219 (discriminator 4)
  #13 0x400f91e6:0x3ffb2530 in MicroOcpp::MOcppMongooseClient::MOcppMongooseClient(mg_mgr*, char const*, char const*, char const*, char const*, std::shared_ptr<MicroOcpp::FilesystemAdapter>, MicroOcpp::ProtocolVersion) at .pio/libdeps/esp32-development-board/MicroOcppMongoose/src/MicroOcppMongooseClient.cpp:76
  #14 0x400d2ed9:0x3ffb25c0 in setup() at src/main.cpp:175
  #15 0x400fba26:0x3ffb2820 in loopTask(void*) at C:/Users/atspc/.platformio/packages/framework-arduinoespressif32@3.20006.221224/cores/esp32/main.cpp:42
**------------------------------------------------------------------------------------------**

**-------------------------------Logs when connected using wifi with **MicroOcppMongoose** adapter-----------------------**
[main] Wait for WiFi: ........[MO] 
warning (MicroOcppMongooseClient.cpp:202): Insecure connection (WS)
[MO] info (MicroOcpp.cpp:349): initialized MicroOcpp v1.1.0
[MO] info (MicroOcppMongooseClient.cpp:474): connection ws://192.168.1.22:1884/fu5v1 -- connected!

[MO] Send: [2,"1000000","BootNotification",{"chargePointModel":"AC","chargePointSerialNumber":"GN234400","chargePointVendor":"myComp","firmwareVersion":"0.0.10"}]
[MO] Recv: [3,"1000000",{"currentTime":"2024-06-11T04:41:41.831724900Z","interval":900,"status":"Accepted"}]
[MO] info (BootNotification.cpp:93): request has been Accepted
[MO] info (StatusNotification.cpp:53): New status: Available (connectorId 0)
[MO] info (StatusNotification.cpp:53): New status: Available (connectorId 1)
[MO] info (StatusNotification.cpp:53): New status: Available (connectorId 2)
[MO] Send: [2,"1000001","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-06-11T04:41:41.843Z"}]
[MO] Recv: [3,"1000001",{}]
[MO] Send: [2,"1000002","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available","timestamp":"2024-06-11T04:41:41.844Z"}]
[MO] Recv: [3,"1000002",{}]
[MO] Send: [2,"1000003","StatusNotification",{"connectorId":2,"errorCode":"NoError","status":"Available","timestamp":"2024-06-11T04:41:41.855Z"}]
[MO] Recv: [3,"1000003",{}]
**------------------------------------------------------------------------------------------**

I have also attched the main file with it, is there any build flag setting i need to do in platform.ini file?

[main.txt](https://github.com/user-attachments/files/15782406/main.txt)

liyong4github, 2024-06-21:
Hi, @matth-x ,@priya2212
I've encountered a similar problem, when using a W5500 to connect to the network, either using mongoose or arduino websockets, the websockets can't connect to the server, the error message is DNS timeout. when using wifi to connect to the network I can connect to the OCPP server correctly.
When using W5500 to connect to the network, I bridge the network to ESP32 softap at the same time, and other devices can access the network normally after connecting to the AP.