matth-x/MicroOcpp #219: Websocket over AT commands for SIM800 or Other GSM modules

FlextronPvtLtd, 2023-10-19:
How dose MicroOcpp connect to WebSocket, where can we see the handshake logs with headers ? We are trying to implement MicroOcpp on sim 800c / 4G module using esp32. 

Did not find any WebSocket libs in the new release either.

matth-x, 2023-10-22:
Some information on how to get started with writing a GSM integration can be found here: https://github.com/matth-x/MicroOcpp/issues/90#issuecomment-1247181869

My advise is to break this project down a little and focus on establishing and maintaining a robust WebSocket connection first. I'm confident that there are some excellent WebSocket integrations for the ESP32 and the SIM800C modem, but unfortunately I don't have a recommendation at hand and this will need some research and evaluation.

Afterwards you can integrate the WebSocket connection into MicroOcpp by subclassing the [`Connection`](https://github.com/matth-x/MicroOcpp/blob/master/src/MicroOcpp/Core/Connection.h) interface. I will publish a more extensive guide on the latter part soon, but please follow up on this issue if you need it earlier.

FlxMojojojo, 2023-10-25:
@matth-x The problem here is, some of the modules are we are use like 800c a7672 etc have there own AT commands and friwmare stack that directly connect to the Websocket, We are able to connect to a normal websocket with our GSM/4G module. And we are stress testing the WebSocket connection to be robust. Right it seems to be stable, we need to now move towards integration of OCCPP library. Can we discuss on this ? We are working with QUALCOMM and simcom together to understands challenges on the module and firmware support.

matth-x, 2023-10-25:
If the modem already provides you with WebSocket connectivity, it's even better. The central concept of the OCPP lib for integrating external WebSocket clients is the [Connection](https://github.com/matth-x/MicroOcpp/blob/master/src/MicroOcpp/Core/Connection.h) interface. A Connection is the communication link between the OCPP lib and the (remote) server. The OCPP lib uses it to send and receive JSON messages and it frequently checks if the link is still up. But it doesn't care about the underlying networking technology. You can write a custom Connection implementation which handles sending and receiving messages by forwarding all messages from and to the Qualcomm and Simcom WebSocket client. To make the OCPP lib use that custom driver, pass the Connection instance to the `mocpp_intialize(...)` initialization function.

Application references for this are the [Mongoose WebSocket adapter](https://github.com/matth-x/MicroOcppMongoose) (this forwards all messages to the Mongoose Networking Library) and the [WebAssembly WebSocket integration](https://github.com/matth-x/MicroOcppSimulator/blob/main/src/net_wasm.cpp).

FlxMojojojo, 2023-11-17:
@matth-x  We defined the custom websocket function ( it disabled the WIFI  and ardunio websocket) in the MOCPP lib, while initializing the OCPP lib, we get an error, OCPP uninitialized,  We think the prams we need pass the connection interface requires on esp wifi and Arduino WebSocket to , How do we pass ? 

Also, Still finding issues on building the connection interface with AT commands, we are able to establish the WebSocket connection with Steve using the module manually. However there are custom AT commands to send and receive the messages. Example AT+SEND "" | While the current lib use send txt and SetRecive txt call back.

feyzanoz, 2023-12-07:
Hello @FlxMojojojo . I'm trying to use MOCPP with the Quectel EC25 modem. What steps should I follow? Can you share the code? Could you please guide me through the necessary stages? How to integrated AT Commands?. Thank you.

feyzanoz, 2023-12-29:
Hi FlxMojojojo, is there any progress?