matth-x/MicroOcpp #222: Unable to connect with Steve using postman

chandan-ultraviolette, 2023-10-25:
I am able to start the SteVe but when I am trying to connect using a postman, I am facing an issue, unable to connect. Sharing the screenshot of my change, any suggestions? I tried connecting with both 192.168 and 10.0 ws url.

[image]

[image]

[image]

Logs are 'No protocol (OCPP version) is specified".

[image]

FlxMojojojo, 2023-10-25:
@chandan-ultraviolette Post your curl request here ? Let me check.

chandan-ultraviolette, 2023-10-25:
@FlxMojojojo Request: ws://192.168.128.230:8080/steve/websocket/CentralSystemService/uv_box_001

FlxMojojojo, 2023-10-25:
@chandan-ultraviolette Please also share the headers that require handshake ?

chandan-ultraviolette, 2023-10-25:
@chandan-ultraviolette  I am not aware what should we send  in header, may be that's the issue.

FlxMojojojo, 2023-10-25:
A WebSocket connection with steve is is an http request first that is sent with headers to upgrade the connection to WebSocket.  Headers play a key role here. Also, While connecting to steve WebSocket service you need to understand the connection request sent by the OCPP charger.

chandan-ultraviolette, 2023-10-25:
Okay, I have not much idea on this, can you help ?

[image]

matth-x, 2023-10-25:
WebSockets have a protocol attribute. Most generic WebSocket clients contain a free-text input field for the protocol specifier. The value must be "ocpp1.6" then.

FlxMojojojo, 2023-10-25:
@chandan-ultraviolette For you reference - Ps. You are in indiranagar we are in HSR ;) 
 
[image]

closed 2023-10-27 by chandan-ultraviolette

chandan-ultraviolette, 2023-10-31:
Btw I am hosting Steve on PC, which setting do I need to change to connect it from any network. Currently unable to connect from other network