matth-x/MicroOcpp #223: SetChargingProfile reports not implemented error;

Monika-18mis, 2023-10-27:
Hi @matth-x 

I have been working on setchargingprofile. After connecting to the server when I send setchargingprofile from server.  It is reporting the below error message.  This is not entering the setchargingprofile.cpp file and it is also not entering smartchargingservice.cpp file.

[OCPP] ERROR (Operation.cpp:38): Unsupported operation: processReq() is not implemented
[OCPP] ERROR (Operation.cpp:42): Unsupported operation: createConf() is not implemented
[OCPP] Send: [4,"685b5bf2-57ab-4916-8a2e-4c6d8caaddff","NotImplemented","",{}]

How can I initialize smartcharging service?
IDE : ESP-IDF 5.1
Board : ESP32 Dev Module

Hope to hear from you soon!

Thanks.

matth-x, 2023-10-27:
Hi @Monika-18mis,

The SmartChargingService is initialized as soon as you set a SmartChargingOutput (see MicroOcpp_c.h).

I must admit that the documentation is not clear about this behavior and I will add a few more details about it.

Monika-18mis, 2023-10-30:
Hi @matth-x, 
If I called the below function and I gave a random value for chargingLimitOutput from main.c .  It works but it is not sending Accepted message, It sends the Rejected message.
ocpp_setSmartChargingOutput(chargingLimitOutput);

What value to be assigned to chargingLimitOutput?

Here are the logs,

[OCPP] Recv: [2,"39f80787-eba1-4423-9987-d34b00b85a1b","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"transactionId":1698613252,"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[OCPP] Send: [3,"39f80787-eba1-4423-9987-d34b00b85a1b",{"status":"Rejected"}]
[OCPP] info (SmartChargingService.cpp:522): New limit for connector 1, scheduled at = 2023-10-30T10:37:20.882Z, nextChange = 2037-01-01T00:00:00.000Z, limit = {-1.000000,-1.000000,-1}

Thanks.

matth-x, 2023-10-30:
I must admit that the log messages should explain the rejection by default. Can you increase the log level from DL_INFO to DL_DEBUG (in the root CMakeLists.txt)?

Monika-18mis, 2023-10-30:
What value to be assigned to chargingLimitOutput?

Do I need to assign the value of current or power in Watts or Amps?

Is it because of the value assigned to chargingLimitOutput?

Thanks.

matth-x, 2023-10-31:
The SmartChargingOutput seems right here. Either Watts or Amps, or the combined Output with Watts, Amps and number of phases is good.

I'm wondering if the charging profile is valid in this scenario. For example it specifies a transactionId which won't go through if the active transaction is different. By increasing the log level you may get more insights to this.

Monika-18mis, 2023-11-20:
Hi @matth-x 

Smart charging is working fine.  And at the time of schedule it is sending this log from the client side.  
What does it do in the charging period? 
I understand that it changes to the MAX_TIME when charging is completed.  
How can I intimate that charging is completed? 
Is there any Implementation for such case?  

info (SmartChargingService.cpp:576): New limit for connector 1, scheduled at = 2023-11-20T09:22:13.651Z, nextChange = 2023-11-20T09:24:00.000Z, limit = {-1.000000,-1.000000,-1}

Thank you.

qombajn, 2024-04-25:
Hi All, how to implemented charging profile?
typedef struct 
{
    float power; 
    float current; 
    int nphases;
}_charging_profile;

_charging_profile CHRG_charging_profile_1;

ocpp_setSmartChargingOutput_m(1,(OutputSmartCharging_m)&CHRG_charging_profile_1);

With that implementation I always crash... How to do it? 
BTW. I want only one connector available in the charge but with ocpp_setSmartChargingOutput (without giving connector ID) it doesn't work too...

qombajn, 2024-04-25:
OutputSmartCharging_m CHRG_charging_profile_1;

ocpp_setSmartChargingOutput_m((unsigned int)1,CHRG_charging_profile_1);

Doesn't work too

qombajn, 2024-04-25:
/*BEFORE TASK*/
void CHRG_charging_profile(unsigned int connectorId, float power, float current, int nphases) {
    printf("NEW_PROFILE_SET\n");
}

/*IN TASK*/
ocpp_initialize(osock, CHARGER_NAME, BRAND_NAME, fsopt, false);
ocpp_setSmartChargingOutput_m((unsigned int)1,&CHRG_charging_profile);

Gives me that logs:

[MO] Recv: [2,"7c5e20d8-f0f1-40fe-977b-d89bcfc8c2b3","SetChargingProfile",{"connectorId":0,"csChargingProfiles":{"chargingProfileId":392080142,"stackLevel":0,"chargingProfilePurpose":"ChargePointMaxProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-04-25T22:26:51Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":10.0,"numberPhases":3}]}}}]
        
[MO] Send: [3,"7c5e20d8-f0f1-40fe-977b-d89bcfc8c2b3",{"status":"Accepted"}]
NEW_PROFILE_SET

[MO] info (SmartChargingService.cpp:510): New limit for connector 1, scheduled at = 2024-04-25T22:27:13.560Z, nextChange = 2037-01-01T00:00:00.000Z, limit = {-1.000000,10.000000,3}

I'm sending ChargePointMaxProfile and ConnectorId = 0, but works somehow. I don't understand yet how profiles works but may be useful.